ZoneMinder/zoneminder #3656: Recording pre-alarm frames are missing

maddios, 2023-01-26:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
zm version, 1.36.33 (zoneminder_1.36.33~20230126.20-bionic_amd64.deb)
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
installed by building from source via OS=ubuntu DIST=bionic utils/packpack/startpackpack.sh then dpkg the resulting build .deb file
- Full name and version of OS
Ubuntu Bionic 18.04 LTS
- Browser name and version (if this is an issue with the web interface)
Chrome 108.0.5359.125 (windows 11)

**Describe the bug**
The issue is since upgrading to 1.36.33 (last night) I noticed all my recorded videos have no frames until the first alarm frame. This was working fine with 1.36.32 (though I was having a separate login issue)
I have it setup to record 100 pre-alarm frames and 100 post-alarm frames. Those 100 pre-alarm frames are frozen and slightly garbled up, I can scrub within the 100 frames and the video is frozen (my camera records sound and I do hear the sound). As soon as the first alarm frame is hit the video is fine that, even any subsequent post alarm frames and secondary alarm frames.

Which are the relevant debug logs I should grab for this (zms?)?

**Debug Logs**

```

maddios, 2023-01-26:
Oh, another thing I just noticed, not sure if it's related but likely.

My camera stream is 5 fps, but the event list shows these bad recordings with WAY too many frames. 

example: video duration 1 minute 40 seconds, but it's showing 1920 frames. Should be ~500.

After rolling back to .32 I'm again seeing sane frame counts.

connortechnology, 2023-02-08:
Are the recordings done using passthrough? or jpegs?

Would need zmc logs at debug level 4.

maddios, 2023-02-08:
These camera is setup as passthrough, jpeg frames disabled.

I'll try to reproduce this again and get you debug logs.

commit 4889358f (2023-02-13): Add the actual packet, not the first packet over and over and over. Fixes #3656
closed 2023-02-13 by connortechnology