ZoneMinder/zoneminder #3658: interactive intercom function between the monitoring center and the front-end camera?

ggsnpdsn, 2023-02-08:
How to realize the interactive intercom function between the monitoring center and the front-end camera? Is this function available at present?

connortechnology, 2023-02-08:
This is not implemented.  It would be a fair amount of work to implement.  Are you interested in funding it?

What is required is two things: 
1 Implementation of webrtc technology in the monitoring center
2 Discovery and implementation of camera api to achieve audio output.

ggsnpdsn, 2023-02-13:
> This is not implemented. It would be a fair amount of work to implement. Are you interested in funding it?
> 
> What is required is two things: 1 Implementation of webrtc technology in the monitoring center 2 Discovery and implementation of camera api to achieve audio output.

Is it necessary to implement freeswitch or asterisk server on the server, and then implement webrtc on the front end? Do you have a general idea of how to integrate this function? I'm interested in trying

baudneo, 2023-04-20:
I am gearing up to try and add 2 way audio using WebRTC (for things like a doorbell but, any 2 way audio camera should work), stay tuned.