ZoneMinder/zoneminder #3660: Janus Live Stream no video "No iceServers defined"

iophi, 2023-02-15:
**Describe Your Environment**
- Version of ZoneMinder: v1.37.31, upgraded from 1.36.32
- How you installed ZoneMinder: ppa:iconnor/zoneminder-1.36
- Full name and version of OS: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
- Browser name and version (if this is an issue with the web interface): Chrome 110.0.5481.100

- Make and Model: Hikvision b4
- frame rate: 6
- resolution: 2560x1440 1440p QHD
- ZoneMinder Source Type: rtsp

**Describe the bug**
While "Janus Live Stream" is enabled under Viewing tab, at the view page of a camera output there's a
gray, default styled html player not playing anything. In the browser's console theres an error:
```
Error: No iceServers defined
Janus @ module.js:1
callback @ js_MonitorStream-base-1676477110.js:192
Janus.init @ module.js:1
MonitorStream.start @ js_MonitorStream-base-1676477110.js:191
initPage @ skins_classic_views_js_watch-base-1676477110.js:734
e @ skins_classic_js_jquery.min-base-1676477110.js:2
t @ skins_classic_js_jquery.min-base-1676477110.js:2
setTimeout (async)
(anonymous) @ skins_classic_js_jquery.min-base-1676477110.js:2
c @ skins_classic_js_jquery.min-base-1676477110.js:2
fireWith @ skins_classic_js_jquery.min-base-1676477110.js:2
fire @ skins_classic_js_jquery.min-base-1676477110.js:2
c @ skins_classic_js_jquery.min-base-1676477110.js:2
fireWith @ skins_classic_js_jquery.min-base-1676477110.js:2
ready @ skins_classic_js_jquery.min-base-1676477110.js:2
$ @ skins_classic_js_jquery.min-base-1676477110.js:2
```
**To Reproduce**
1. Install Ubunt
2. Install Zoneminder from that ppa
3. Follow the installation steps from the docs
4. Upgrade the package
5. Setup cameras
6. install Janus
7. Configure janus and janus related zoneminder options in the options->system
8. Visit zoneminder web panel camera view from a pc on the same network and get this error

I followed [this guy's steps](https://www.youtube.com/watch?v=otAVgAMSUB8) of configuring janus and setting it up.
From what I've searched, no one had written about this issue to this moment. 

[image]

**My setup**
Wired Linux box on home wifi network, POE switch and IP cameras.

- /etc/janus/janus.jcfg

```
general: {
       general: {
        configs_folder = "/etc/janus"                   # Configuration files folder
        plugins_folder = "/usr/lib/x86_64-linux-gnu/janus/plugins"                     # Plugins folder
        transports_folder = "/usr/lib/x86_64-linux-gnu/janus/transports"       # Transports folder
        events_folder = "/usr/lib/x86_64-linux-gnu/janus/events"                       # Event handlers folder
        loggers_folder = "/usr/lib/x86_64-linux-gnu/janus/loggers"                     # External loggers folder

        log_to_file = "/var/tmp/janus.log"              # Whether to use a log file or not
        debug_level = 3                                       # Debug/logging level, valid values are 0-7
        protected_folders = [
                "/bin",
                "/boot",
                "/dev",
                "/etc",
                "/initrd",
                "/lib",
                "/lib32",
                "/lib64",
                "/proc",
                "/sbin",
                "/sys",
                "/usr",
                "/var",
                "/opt/janus/bin",
                "/opt/janus/etc",
                "/opt/janus/include",
                "/opt/janus/lib",
                "/opt/janus/lib32",
                "/opt/janus/lib64",
                "/opt/janus/sbin"
        ]
}

certificates: {
}

media: {}

nat: {
        nice_debug = false
        ignore_mdns = true        
        ice_ignore_list = "vmnet"

}

plugins: {
}

transports: {
}

loggers: {
}

events: {
}

```
- /etc/janus/janus.plugin.streaming.jcfg
```
general: {
        admin_key = "my_secret"             # If set, mountpoints can be created via API

}
```
- /etc/janus/janus.plugin.streaming.jcfg
```
JANUS_SECRET = my_secret
JANUS_PATH  = http://192.168.0.150:8088/janus
```
- /var/tmp/janus.log
```
Logger plugins folder: /usr/lib/x86_64-linux-gnu/janus/loggers
Loading logger plugin 'libjanus_jsonlog.so'...
[WARN] JSON logger disabled
---------------------------------------------------
  Starting Meetecho Janus (WebRTC Server) v0.11.8
---------------------------------------------------

Checking command line arguments...
Debug/log level is 3
Debug/log timestamps are disabled
Debug/log colors are disabled
[WARN] mDNS resolution disabled, .local candidates will be ignored
[WARN] Janus is deployed on a private address (192.168.0.150) but you didn't specify any STUN server! Expect trouble if this is supposed to work over the internet and not just in a LAN...
[WARN] RabbitMQ support disabled for both Janus and Admin API, giving up
[WARN] The 'janus.transport.rabbitmq' plugin could not be initialized
[ERR] [config.c:janus_config_parse:205] Error parsing config file at line 30: syntax error
[WARN] Couldn't find .jcfg configuration file (janus.transport.mqtt), trying .cfg
[ERR] [config.c:janus_config_parse:191]   -- Error reading configuration file 'janus.transport.mqtt.cfg'... error 2 (No such file or directory)
[WARN] MQTT support disabled for both Janus and Admin API, giving up
[WARN] The 'janus.transport.mqtt' plugin could not be initialized
[WARN] No Unix Sockets server started, giving up...
[WARN] The 'janus.transport.pfunix' plugin could not be initialized
[ERR] [janus.c:janus_process_incoming_request:1147] Couldn't find any session 1989143777619814...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
```

jp-bennett, 2023-02-15:
> [ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:6457] Couldn't get DESCRIBE code: 401
> [ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...

This is the actual error. My guess is that your version of libcurl has a known bug in it. I think this was fixed in libcurl 7.87.0

iophi, 2023-02-16:
Yup, I had curl version 7.81.0-1ubuntu1.7 installed. I removed it and Intalled curl 7.88.0 form [here](https://curl.se/download.html), now everything works great! I wouldn't have thought that this was the issue. Thank You!

closed 2023-02-16 by iophi