ZoneMinder/zoneminder #3665: Debian Buster - can't update/install ZoneMinder v1.36.x via zmrepo

JedMeister, 2023-03-01:
**Describe Your Environment**
- Version of ZoneMinder: current: `1.34.26-buster` - trying to install: `1.36.33-buster1`
- How you installed ZoneMinder: `apt` via zmrepo.zoneminder.com
- Full name and version of OS: Debian 10/Buster
- Browser name and version: N/A

**Describe the bug**
Trying to upgrade ZoneMinder (on a Debian Buster system) from v1.34.x to v1.36.x fails due to unfulfilled dependencies:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 zoneminder : Depends: libavcodec58 (>= 7:4.4)
              Depends: libavformat58 (>= 7:4.4) but 7:4.1.10-0+deb10u1 is to be installed
              Depends: libavutil56 (>= 7:4.4) but 7:4.1.10-0+deb10u1 is to be installed
              Depends: libc6 (>= 2.34) but 2.28-10+deb10u2 is to be installed
              Depends: libgcc-s1 (>= 3.3.1) but it is not installable
              Depends: libgcrypt20 (>= 1.9.0) but 1.8.4-5+deb10u1 is to be installed
              Depends: libgnutls30 (>= 3.7.0) but 3.6.7-4+deb10u10 is to be installed
              Depends: libjpeg8 (>= 8c) but it is not installable
              Depends: libmysqlclient21 (>= 8.0.11) but it is not installable
              Depends: libstdc++6 (>= 12) but 8.3.0-6 is to be installed
              Depends: libswresample3 (>= 7:4.4) but 7:4.1.10-0+deb10u1 is to be installed
              Depends: libswscale5 (>= 7:4.4) but 7:4.1.10-0+deb10u1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

**To Reproduce**
Steps to reproduce the behavior:
1. On a Debian Buster system - configure the 1.36.x zmrepo
2. run `apt update` & `apt install zoneminder`

**Expected behavior**
Expected ZoneMinder v1.36.x apt install to "just work".

I'm only guessing, but I can't help but wonder if perhaps a package intended for Ubuntu has inadvertently been uploaded to the Buster repo?!

**Debug Logs**
I downloaded and compared the debs for both Buster and Bullseye and checked their control files.

Before I note what I found there, it's perhaps also worth noting that the internals of the Buster package use zst compression - which is not supported on Buster (it isn't even supported on Bullseye, but should be in Bookworm). So I suspect that even if the dependencies were available, install would still fail. The Bullseye package uses xz (which should work fine on Buster too).

The Bullseye one has dependencies that are fulfill-able from default Debian Bullseye repos (as expected). Although the thing that strikes me is that many of the declared dependencies are actually newer than the Bullseye ones!?!

`zoneminder_1.36.33-buster1_amd64.deb`:
```
Package: zoneminder
Version: 1.36.33-buster1
Architecture: amd64
Maintainer: Isaac Connor <isaac@zoneminder.com>
Installed-Size: 32256
Depends: libavcodec58 (>= 7:4.4), libavformat58 (>= 7:4.4), libavutil56 (>= 7:4.4), libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), libgcrypt20 (>= 1.9.0), libgnutls30 (>= 3.7.0), libjpeg8 (>= 8c), libjwt-gnutls0 (>= 1.9.0), libmysqlclient21 (>= 8.0.11), libpcre3, libstdc++6 (>= 12), libswresample3 (>= 7:4.4), libswscale5 (>= 7:4.4), zlib1g (>= 1:1.1.4), perl:any, javascript-common, ffmpeg, libdate-manip-perl, libmime-lite-perl, libmime-tools-perl, libdbd-mysql-perl, libphp-serialization-perl, libmodule-load-conditional-perl, libnet-sftp-foreign-perl, libarchive-zip-perl, libdevice-serialport-perl, libimage-info-perl, libio-interface-perl, libjson-maybexs-perl, libsys-mmap-perl, liburi-encode-perl, libwww-perl, liburi-perl, libdata-dump-perl, libdatetime-perl, libclass-std-fast-perl, libsoap-wsdl-perl, libio-socket-multicast-perl, libdigest-sha-perl, libsys-cpu-perl, libsys-meminfo-perl, libdata-uuid-perl, libnumber-bytes-human-perl, libfile-slurp-perl, default-mysql-client | mariadb-client | virtual-mysql-client, php-mysql, php-gd, php-apcu, php-json, php-intl, policykit-1, rsyslog | system-log-daemon, zip, libcrypt-eksblowfish-perl, libdata-entropy-perl, libvncclient1 | libvncclient0
Recommends: apache2 (>= 2.4.6-4~) | nginx | httpd, libapache2-mod-php | php-fpm, default-mysql-server | mariadb-server | virtual-mysql-server, zoneminder-doc (>= 1.36.33-buster1)
Suggests: fcgiwrap, logrotate
Section: net
Priority: optional
Homepage: https://www.zoneminder.com/
Description: video camera security and surveillance solution
 ZoneMinder is intended for use in single or multi-camera video security
 applications, including commercial or home CCTV, theft prevention and child
 or family member or home monitoring and other care scenarios. It
 supports capture, analysis, recording, and monitoring of video data coming
[... 50 lines cut ...]
 *** 7:4.1.10-0+deb10u1 500
        500 http://security.debian.org buster/updates/main amd64 Packages
        100 /var/lib/dpkg/status
     7:4.1.9-0+deb10u1 500
        500 http://deb.debian.org/debian buster/main amd64 Packages

libavformat58:
  Installed: 7:4.1.10-0+deb10u1
  Candidate: 7:4.1.10-0+deb10u1
  Version table:
 *** 7:4.1.10-0+deb10u1 500
        500 http://security.debian.org buster/updates/main amd64 Packages
        100 /var/lib/dpkg/status
     7:4.1.9-0+deb10u1 500
        500 http://deb.debian.org/debian buster/main amd64 Packages

libgcc-s1:
  Installed: (none)
  Candidate: (none)
  Version table:
```

After a little more research, it seems that the `/lib/x86_64-linux-gnu/libgcc_s.so.1` file provided by the `libgcc-s1` package on Bullseye is provided by the `libgcc1` package on Buster (current version is [`1:8.3.0-6`](https://packages.debian.org/buster/libgcc1)).

Unfortunately though, I'm not completely sure whether those files are interchangeable (I suspect so - but I'm only guessing).

connortechnology, 2023-03-02:
It should just work.  I'll do some testing.

JedMeister, 2023-03-02:
I can adjust those dependencies for the Buster package by hand and test to see if it runs ok if that's of any value?

JedMeister, 2023-03-02:
FWIW v1.34.x upgrades to v1.36.32 no problem. I.e. this works as expected:
```
apt install zoneminder=1.36.32-buster1
```

So it appears to just be an issue with the v1.36.33 build.

As an aside, the `zoneminder.service` appears to be in a failed-restart loop -  although Zoneminder appears to be running fine?! This is easily resolved by stopping the service (`systemctl stop zoneminder`), killing the processes (`kill -9 $(pgrep zm)`), then uncommenting the 'User=www-data' line in the service file (`sed -i "\|^#User|s|^#||" /usr/lib/systemd/system/zoneminder.service`), then finally starting the service again (`systemctl start zoneminder`).

FWIW, I found mention of that workaround a few places online so it appears that I'm not completely alone in hitting that issue, but I couldn't find enough mention of it (including no bug here) to think that it is really widespread?!

Is there a rational for having that commented by default? Or is there something not quite right with my default permissions somewhere? Or something else I haven't considered? Should I open a separate issue? (So many questions... :grin:)

aval13, 2023-03-12:
I have the same issue on Buster with 1.36.33, .32 works just fine, I fully agree with @JedMeister finds.
In my case, so I don't change the official unitfile which came with the package (bad practice to mess with package provided files), my workaround was to create /etc/systemd/system/zoneminder.service.d/userFix.conf which only says:
[Service]
User=www-data
Did that so long ago I forgot it was there.
My best guess on my memories is that since zoneminder is "integrated" with apache (www-data user on Debian) and if you use the web interface, the web gui (running inside apache as www-data) will need to access files on disk (recordings etc) written by the zm* processes, so in order to "unify" the setup makes sense to run zoneminder as www-data so that the web works normally.
The other option would be to:
- run zoneminder as a dedicated user (running as root is bad security practice)
- put that user in the www-data group
- setup the data folder of zoneminder (where is saves the frames, videos etc as 2750 owned by www-data group)

Too many steps and my guess at this hour (2:25AM) is that it will also cause issues when trying to delete event data on disk (maybe 2770 would work?).
So, yeah, the run-as www-data is the simplest solution.

JedMeister, 2023-04-05:
Thanks @connortechnology - I don't have a buster system handy right now, but I did have a quick look at the control file and it looks good to me. So I'll shut this now. Thanks again.

closed 2023-04-05 by JedMeister