ZoneMinder/zoneminder #3666: No time display on events

basildane, 2023-03-02:
**Describe Your Environment**
- Version of ZoneMinder - 1.36.33
- How you installed ZoneMinder - PPA
- Full name and version of OS - Ubuntu Server
- Browser name and version - Brave

**If the issue concerns a camera**
N/A

**Describe the bug**
The recorded event viewer used to display the channel name and time on at the bottom of the video.
This was working in 1.36.32.  Today I updated to 1.36.33 and the time display is gone.  When you play an event, it's just raw video with no indication of what time it occurred.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Console
2. Click on an Event
3. See video - no time display

**Expected behavior**
Playing an event, need to see the time on the video

**Debug Logs**
```

```

connortechnology, 2023-03-02:
See #3488 

We are going with the stance that we should not be overlaying things over the video that may block seeing useful parts of video.

basildane, 2023-03-02:
OK, glad there is a reason for this and I can stop troubleshooting it.

So I want to ask your advice on how to proceed, because for any security uses (which is what I need zoneminder for), having the exact time burned in the video is absolutely required.
I am using pass-through video at the moment, because all the HD cameras were overloading the machine.

Do you have any recommendations?

connortechnology, 2023-03-02:
Have the cameras do it.  Passthrough has no hard coded time unless the cameras do it.  That little display meant nothing.  It was simply generated from the db record.

hansrijn, 2023-03-03:
Migrating from Raspberry Pi version 1.34.22 to Ubuntu v1.36.33 gave me the same "problem". I liked the 1.34.22 situation better. Could it not be optional?

hansrijn, 2023-03-05:
Why I liked it better (and many more will, it saves trouble to get all the camera's at the right time. Some of mine will even refuse to do it right. )
[image]
[image]

gandihar, 2023-03-07:
I also used the overlay timestamp on videos. When reviewing a large (most of the screen) video stream, I don't get the sidebar with the time/date, so I used the overlay time quite a bit. Now it's gone. I have to shrink the video window to get the sidebar to show up, which makes seeing anything inside the video window more difficult.

basildane, 2023-03-25:
I don't think you know what a massive pain in the butt this is for most people.

I have built an NTP server now, put it on my private VLAN to support the camera array, and configured firewall rules for it.
Then I started manually configuring all the cameras.  This simply is not going to work.
I have not encountered even ONE SINGLE camera that properly supports daylight savings time, and some cameras cannot even use my NTP server.  They are hard coded to go to nist.gov or something, which is impossible because they are not on the internet.

Therefore, I am now forced to have no time stamping in security recordings.

Please reconsider this functionality.

connortechnology, 2023-03-25:
If you want it back, edit /usr/share/zoneminder/www/skins/classic/css/base/views/event.css and delete the last three lines. 

But you need to be aware, there is NO TIMESTAMP IN YOUR VIDEO unless you put it there, either in the camera, or by re-encoding the video.  Your video will be useless as evidence.

I know exactly what a pain in the ass it is to manage all this.  Which is why people pay for cloud cameras.

sanddastan, 2024-12-13:
Man, here i am at 3:30am migrating my 1.34 to 1.36 ZM install and I ran in this exact problem! Seriously, I really love and appreciate zoneminder and the dev work but....you guys shouldn't be changing things like this, at least make it a separate UI configuration. Thank you for pointing out how to fix this however!

This was a main issue I ran into, and need to remember to manually edit this file going forward if I update again, and another issue is the screen cut off/can't zoom out when on mobile browser that did not happen on previous versions.

Here's the thing, I am not a fan of updating, but I figured I must be a crazy person still using 1.34 on an old version of Ubuntu so decided to update everything and I am 50/50 on it. Next time once I get all of the kinks out i don't think ill bother updating it again unless it breaks because I will have to re-learn things.

Editing /usr/share/zoneminder/www/skins/classic/css/base/views/event.css worked for me though. Many of us use camera pass through for performance, sometimes the CCTVs do not let you set your own NTP server, or their time display uses up a lot of the screen so we prefer to just have it when viewing the event. We can screen record that and still use that as evidence. but thank you nonetheless.

closed 2026-06-12 by SteveGilvarry