ZoneMinder/zoneminder #3670: ip command not found

AspireONE-zz, 2023-03-06:
on Fedora 37 'ip' command is located in /usr/sbin/ip. 
When I run ONVIF probe it does not find network interfaces and returns error 127.

doing : ln -s /usr/sbin/ip /usr/bin/ip

confirms that the problem is z search patch of zm

**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
- Full name and version of OS
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

knight-of-ni, 2023-03-07:
You neglected to state how you installed zoneminder on f37. It matters. 

Cmake searches for a compatible ARP executable during build:
https://github.com/ZoneMinder/zoneminder/blob/master/CMakeLists.txt#L653

Specifically, the ZoneMinder rpm from RPMFusion doesn't use the `ip` executable. It uses the `arp` executable by pulling in the net-tools package during the build:
https://github.com/ZoneMinder/zoneminder/blob/master/distros/redhat/zoneminder.spec#L59

Post the contents of `/etc/zm/conf.d/01-system-paths.conf` on your system.

AspireONE-zz, 2023-03-07:
Hello,

I have pulled the package from RPMFusion (as you have guessed).

%cat /etc/zm/conf.d/01-system-paths.conf | grep ^ZM

ZM_DIR_EVENTS=/var/lib/zoneminder/events
ZM_DIR_SOUNDS=sounds
ZM_DIR_EXPORTS=/var/lib/zoneminder/temp
ZM_PATH_ZMS=/cgi-bin-zm/nph-zms
ZM_PATH_MAP=/dev/shm
ZM_PATH_SOCKS=/var/lib/zoneminder/sock
ZM_PATH_LOGS=/var/log/zoneminder
ZM_PATH_SWAP=/var/lib/zoneminder/temp
ZM_PATH_ARP="/usr/sbin/arp"
ZM_PATH_SHUTDOWN="/sbin/shutdown"
ZM_PATH_FFMPEG="/usr/bin/ffmpeg"

% rpm -qif /etc/zm/conf.d/01-system-paths.conf | grep RPM

Source RPM  : zoneminder-1.36.32-1.fc37.src.rpm
Packager    : RPM Fusion

HOWEVER, from /var/log/zoneminder/web_php.log

3/6/23, 4:59:30 AM UTC.623130 web_php[1531].ERR [192.168.8.148] [Unable to list network interfaces, status is '127'. Output was:<br/><br/>] at /usr/share/zoneminder/www/skins/classic/views/onvifprobe.php line 183
3/6/23, 4:59:30 AM UTC.631956 web_php[1531].ERR [192.168.8.148] [Unable to list network interfaces, status is '127'. Output was:<br/><br/>] at /usr/share/zoneminder/www/skins/classic/views/onvifprobe.php line 199

AND, from  +183 +199 /usr/share/zoneminder/www/skins/classic/views/onvifprobe.php 

  exec('ip route', $output, $status);

  exec('ip link', $output, $status);

knight-of-ni, 2023-03-07:
Thank you. That's the output needed to properly troubleshoot this.

@connortechnology This issue was introduced in 074e270e795d005c5e1c8c11d50c661d97f9081d when the `ip` executable was hardcoded. We know from experience hard coding executables is problematic across distros. Recall all the trouble we had trying to use the 'arp' executable several years ago when it too was hardcoded... using `type` or `which` to find the desired executable path dynamically does not work in all cases either.

If the `ip` executable is absolutely needed, then the fool proof method seems to be to have cmake verify the full path during the build, just like we are are doing for the `arp` executable here: https://github.com/ZoneMinder/zoneminder/blob/master/CMakeLists.txt#L653

connortechnology, 2023-03-07:
Agreed.  Ineresting that it works here on ubuntu, the path must include /usr/sbin, which would surprise me... It is safer to be explicit anyways.

commit 7060945e (2023-09-29): Use new ZM_PATH_IP and ZM_PATH_IFCONFIG instead of hard coded paths. Fixes #3670
closed 2023-09-29 by connortechnology