ZoneMinder/zoneminder #3671: Event Lists View Constantly Resets

gandihar, 2023-03-07:
**Describe Your Environment**
- Version of ZoneMinder: zoneminder-1.36.33-1.el7.x86_64
- Installed From: RPMFusion
- OS: CentOS Linux-7
- Browser: Firefox, Chrome; both current updates

**Describe the bug**
Previous versions of Zoneminder kept state of the Events List. The 1.36.33 update seems to have given the Events List amnesia. If I change the item limit, or subtract columns, it forgets the next time I go into the list, resetting to 10 items and bringing back removed columns.

**To Reproduce**
1. Bring up the Events List for any camera or group of cameras.
2. See the new "Archived" and "Emailed" columns, also that the page limit is 10 items.
3. Change anything. Literally anything. Turn off columns, change row count, etc. Changes are immediate.
4. Click on an event to view.
5. When coming back (either with the back button or navigation) to this Event List and any change made has been undone. Any removed column is back and the page limit is once again 10.

**Expected behavior**
It used to remember at least the row counts. I didn't need to remove columns before, but now it's getting so wide that it's making the list longer as rows expand due to line wrapping. And I don't need "emailed" and "archived" under the event name *and* in its own column.

** Notes **
My apologies if this is a duplicate. I tried searching but did not find an equivalent issue.

robot468, 2023-04-17:
I'm experiencing the same https://github.com/ZoneMinder/zoneminder/issues/3686

krlgithub, 2023-05-16:
me too.  not a crisis,, but would love settings to persist again.

closed 2023-07-12 by connortechnology

krlgithub, 2023-07-16:
@OscarParzon , you rock.  thanks.  simple, clean,  works great.

PR #3726 (2023-07-12): Remember Events table settings
Fixes #3671