ZoneMinder/zoneminder #3674: 'zmc -m 1' exited abnormally, exit status 11

MDBrown01, 2023-03-17:
**Describe Your Environment**
- Version of ZoneMinder [release version, v1.36.33
- How you installed ZoneMinder PPA,
- Full name and version of OS ubuntu-22.04.2-desktop-amd64

**Describe the bug**

Ubuntu Desktop is a new build running on VMWare ESXi 7.02 8vCPU 16GB Ram and 2.5Tb disk space 
I have no errors on install and followed guide to the letter.

TrendNet camera works using VLC Player RTSP stream tested 

3/17/23, 4:16:09 PM UTC | zmdc |   | 14933 | ERR | 'zmc -m 1' exited abnormally, exit status 11
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | INF | Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x55700a027a53 0x7f3ff8c42520 0x55700a00ae70 0x557009fed48d 0x7f3ff8c29d90 0x7f3ff8c29e40 0x557009ff0cc5 | zm_signal.cpp | 108

3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 6: /usr/bin/zmc(+0x18cc5) [0x557009ff0cc5] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f3ff8c29e40] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f3ff8c29d90] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 3: /usr/bin/zmc(+0x1548d) [0x557009fed48d] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 2: /usr/bin/zmc(+0x32e70) [0x55700a00ae70] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f3ff8c42520] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Backtrace 0: /usr/bin/zmc(+0x4fa53) [0x55700a027a53] | zm_signal.cpp | 101
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Signal address is 0x18, from 0x55700a00ae70 | zm_signal.cpp | 80
3/17/23, 4:16:09 PM UTC | zmc_m1 |   | 15967 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
3/17/23, 4:16:02 PM UTC | zmc_m1 |   | 15967 | INF | Starting Capture version 1.36.33 | zmc.cpp | 220
3/17/23, 4:16:02 PM UTC | zmdc |   | 15967 | INF | 'zmc -m 1' started at 23/03/17 16:16:02 | zmdc.pl | -
3/17/23, 4:16:02 PM UTC | zmdc |   | 14933 | INF | 'zmc -m 1' starting at 23/03/17 16:16:02, pid = 15967 | zmdc.pl | -
3/17/23, 4:16:02 PM UTC | zmdc |   | 14933 | INF | Starting pending process, zmc -m 1

These errors are continuous and the same.

If you let me know what and how to get any logs I will send them to you.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console added a camera and set to record'
2. Go To Log and seen these errors

**Expected behavior**
No Camera image displayed 

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2023-03-17:
Go to Options->Logging Set ZM_DEBUG to checked, LEVEL to 4. Restart zoneminder.  Now /var/log/zm/zmc_m1.log should have tons of debugging info.

MDBrown01, 2023-03-18:
Here is the the file from the change to log level 4
I get nothing on the monitor 

03/18/23 09:39:57.012556 zmc_m1[24507].DB1-zm_logger.cpp/224 [LogOpts: level=DB4 effective=DB4, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zmc_m1.log, syslog=DB9]
03/18/23 09:39:57.012631 zmc_m1[24507].DB1-zm_utils.cpp/198 [Detected a x86\x86-64 processor with AVX2]
03/18/23 09:39:57.013023 zmc_m1[24507].DB4-zm_image.cpp/396 [Blend: Using SSE2 fast blend function]
03/18/23 09:39:57.013050 zmc_m1[24507].DB4-zm_image.cpp/458 [Delta: Using SSSE3 delta functions]
03/18/23 09:39:57.013064 zmc_m1[24507].DB4-zm_image.cpp/559 [Deinterlace: Using standard functions]
03/18/23 09:39:57.013075 zmc_m1[24507].DB4-zm_image.cpp/572 [Image buffer copy: Using standard memcpy]
03/18/23 09:39:57.013369 zmc_m1[24507].DB1-zm_storage.cpp/91 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
03/18/23 09:39:57.013394 zmc_m1[24507].DB1-zm_monitor.cpp/493 [Have camera type Ffmpeg]
03/18/23 09:39:57.013406 zmc_m1[24507].DB1-zm_monitor.cpp/2212 [Reloading linked monitors for monitor Front of the House, '(null)']
03/18/23 09:39:57.013421 zmc_m1[24507].DB3-zm_monitor.cpp/570 [Decoding enabled: 1 function 4 Record savejpegs 3 videowriter 0]
03/18/23 09:39:57.013434 zmc_m1[24507].DB1-zm_packetqueue.cpp/714 [Setting pre_event_video_packet_count to 5]
03/18/23 09:39:57.013446 zmc_m1[24507].DB1-zm_packetqueue.cpp/708 [Setting max_video_packet_count to 0]
03/18/23 09:39:57.013458 zmc_m1[24507].DB1-zm_monitor.cpp/658 [mem.size(8) SharedData=760 TriggerData=560 VideoStoreData=4128 timestamps=48 images=3x11943936 = 35831808 total=35837368]
03/18/23 09:39:57.013480 zmc_m1[24507].DB2-zm_camera.cpp/68 [New camera id: 1 width: 2304 line size: 9216 height: 1296 colours: 4 subpixelorder: 8 capture: 1]
03/18/23 09:39:57.013503 zmc_m1[24507].INF-zm_ffmpeg.cpp/84 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
03/18/23 09:39:57.013574 zmc_m1[24507].DB3-zm_monitor.cpp/2204 [Reloading zones for monitor Front of the House have 0]
03/18/23 09:39:57.013891 zmc_m1[24507].DB1-zm_zone.cpp/839 [Got 1 zones for monitor Front of the House]
03/18/23 09:39:57.013919 zmc_m1[24507].DB3-zm_zone.cpp/765 [Got coordinate 0,0 from polygon string]
03/18/23 09:39:57.013936 zmc_m1[24507].DB3-zm_zone.cpp/765 [Got coordinate 2303,0 from polygon string]
03/18/23 09:39:57.013947 zmc_m1[24507].DB3-zm_zone.cpp/765 [Got coordinate 2303,1295 from polygon string]
03/18/23 09:39:57.013959 zmc_m1[24507].DB3-zm_zone.cpp/765 [Got coordinate 0,1295 from polygon string]
03/18/23 09:39:57.013970 zmc_m1[24507].DB3-zm_zone.cpp/777 [Successfully parsed polygon string 0,1295]
03/18/23 09:39:57.014927 zmc_m1[24510].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1679132397.013503, 'zmc_m1', 0, 24507, 0, 'INF', 'Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options', 'zm_ffmpeg.cpp', 84 )]
03/18/23 09:39:57.016476 zmc_m1[24507].DB1-zm_image.cpp/2486 [vertices 4]
03/18/23 09:39:57.019067 zmc_m1[24507].DB1-zm_monitor.cpp/2206 [Reloading zones for monitor Front of the House have 1]
03/18/23 09:39:57.019104 zmc_m1[24507].DB1-zm_monitor.cpp/694 [Loaded monitor 1(Front of the House), 1 zones]
03/18/23 09:39:57.019127 zmc_m1[24507].DB2-zmc.cpp/217 [1 monitors loaded]
03/18/23 09:39:57.019145 zmc_m1[24507].INF-zmc.cpp/220 [Starting Capture version 1.36.33]
03/18/23 09:39:57.019178 zmc_m1[24507].DB3-zm_monitor.cpp/917 [Connecting to monitor.  Purpose is 1]
03/18/23 09:39:57.019212 zmc_m1[24507].DB3-zm_monitor.cpp/930 [Success opening mmap file at (/dev/shm/zm.mmap.1)]
03/18/23 09:39:57.019233 zmc_m1[24507].DB3-zm_monitor.cpp/963 [MMap file size is 0]
03/18/23 09:39:57.020017 zmc_m1[24510].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1679132397.019145, 'zmc_m1', 0, 24507, 0, 'INF', 'Starting Capture version 1.36.33', 'zmc.cpp', 220 )]
03/18/23 09:39:57.064035 zmc_m1[24507].DB3-zm_monitor.cpp/1004 [Aligning shared memory images to the next 64 byte boundary]
03/18/23 09:39:57.064087 zmc_m1[24507].DB3-zm_monitor.cpp/1014 [Allocated 3 3 image buffers]
03/18/23 09:39:57.071553 zmc_m1[24507].DB3-zm_monitor.cpp/1067 [Success connecting]
03/18/23 09:39:57.072685 zmc_m1[24507].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO Monitor_Status (MonitorId,Status,CaptureFPS,AnalysisFPS) VALUES (1, 'Running',0,0) ON DUPLICATE KEY UPDATE Status='Running',CaptureFPS=0,AnalysisFPS=0]
03/18/23 09:39:57.072714 zmc_m1[24507].DB1-zm_ffmpeg_camera.cpp/185 [Priming capture from rtsp://192.168.200.22:554/streaming/Channels/101]
03/18/23 09:39:57.072753 zmc_m1[24507].DB1-zm_ffmpeg_camera.cpp/311 [Calling avformat_open_input for rtsp://192.168.200.22:554/streaming/Channels/101]
03/18/23 09:39:57.072932 zmc_m1[24507].DB3-zm_ffmpeg.cpp/69 [No default whitelist set]
03/18/23 09:39:57.072964 zmc_m1[24507].DB3-zm_ffmpeg.cpp/69 [Original list of addresses:]
03/18/23 09:39:57.072984 zmc_m1[24507].DB3-zm_ffmpeg.cpp/69 [Address 192.168.200.22 port 554]
03/18/23 09:39:57.072996 zmc_m1[24507].DB3-zm_ffmpeg.cpp/69 [Interleaved list of addresses:]
03/18/23 09:39:57.073008 zmc_m1[24507].DB3-zm_ffmpeg.cpp/69 [Address 192.168.200.22 port 554]
[... cut ...]
03/18/23 09:40:04.668396 zmc_m1[24544].DB3-zm_ffmpeg.cpp/69 [nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0]
03/18/23 09:40:04.769084 zmc_m1[24544].DB3-zm_ffmpeg.cpp/69 [nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0]
03/18/23 09:40:04.769117 zmc_m1[24544].DB3-zm_ffmpeg.cpp/69 [nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0]
03/18/23 09:40:04.769131 zmc_m1[24544].DB3-zm_ffmpeg.cpp/69 [nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0]
markbrown@cctv:~$

connortechnology, 2023-03-18:
Okoay, I'm going to dig further, but right off the bat, don't populate the secondary stream url.  That is actually just for an external audio source, not for using the substream.  Eventually it will be used for that, but not yet.

MDBrown01, 2023-03-18:
Okay so i removed the 2nd stream, and I get a green indicator of recording but no image displayed on monitor or on IOS phone live view.

MDBrown01, 2023-03-23:
Any idea whats going on, yet.

Regards

Regards
Mark Brown

MDBrown01, 2023-03-24:
> Probably apache misconfig. Number one problem is that the path to the zms cgi is wrong.

Have you got the correct path so I can compare

Mark

Regards
Mark Brown

SteveGilvarry, 2024-09-13:
Closing as Stale, please reopen if bug, but sounds like config.

closed 2024-09-13 by SteveGilvarry