ZoneMinder/zoneminder #3677: Listing of events page is missing content

yearry, 2023-03-21:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
-      1.37.37
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
-      zmrepo.zoneminder.com/el/8
- Full name and version of OS
-      MacOS Ventura 13.2.1, iOS 16.3.1
- Browser name and version (if this is an issue with the web interface)
-      Safari 16.3, Chrome 111.0.5563.64, iOS Safari Browser

**Describe the bug**
    From the "Console" page clicking on a camera event item that has at least 1 number of events for the specified period (ex. "Hour") shows no items when the page is loaded. This occur for every time period and every device (camera) in the list.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Console"
2. Click on any event time period for any camera (ex. Events, Hour, Day, Week, Month).
3. Takes you to a page that is blank except for the page header/menus.
4. No error is produced only the blank page is displayed.

**Expected behavior**
A clear and concise description of what you expected to happen.
   Previous versions of ZM produced a list of events with date/times, length of event, link to view video, etc.

**Debug Logs**
```
3/21/23, 2:04:37 PM CDT | web_php | 1453 | WAR | Invalid view console in ZM_WEB_HOMEVIEW does not exist at /var/www/html//skins/classic/views/console.php | /usr/share/zoneminder/www/includes/functions.php | 2457
-- | -- | -- | -- | -- | -- | --

```

KameZero, 2023-03-22:
Had the same issue. Looks like the issue is:

`PHP Fatal error:  Uncaught Error: Call to undefined function ZM\\split() in /usr/share/zoneminder/www/includes/Filter.php:1037\nStack trace:\n#0 /usr/share/zoneminder/www/skins/classic/views/events.php(65): ZM\\Filter->simple_widget()\n#1 /usr/share/zoneminder/www/index.php(269): require_once('/usr/share/zone...')\n#2 {main}\n  thrown in /usr/share/zoneminder/www/includes/Filter.php on line 1037, referer: https://zoneminder/index.php?view=console
[Wed Mar 22 10:46:37.362141 2023] [php7:error] `

Changing the split() in Filter.php:1037 to preg_split() shows the events but does not automatically select the monitor. It will show events for all monitors from that timeframe.

yearry, 2023-03-22:
The latest update fixed the issue. Thanks!

yearry, 2023-03-23:
The latest update fixed the issue.

closed 2023-03-23 by yearry