ZoneMinder/zoneminder #3686: web interface not saving sort order of events

robot468, 2023-04-11:
When i change sort order in event list (index.php?view=events), sorting order not saved in cooke.
Cookie value always contains only "sort_asc%5D=1".
Sort order always ascending when i open events view.

Simpler1, 2023-05-12:
This is likely caused by https://github.com/ZoneMinder/zoneminder/commit/8370ae6c82dfb75aa597a0e77aa883e92246c5f9
Try changing
`data-cookie-id-table="zmEventsTable<?php echo $filterQuery?>"`
to
`data-cookie-id-table="zmEventsTable"`
in
`web/skins/classic/views/events.php`

towingie, 2024-08-03:
same here, filter param ASC/DESC (for background working) not being saved to database

SteveGilvarry, 2024-09-13:
Review for 1.38 to see if still present. Any of the posters in this issue please confirm if still seeing this with master builds.

closed 2025-12-29 by SteveGilvarry