ZoneMinder/zoneminder #3687: Could not open codec. Error code -542398533 when live feeding mp4

nejcmedved, 2023-04-15:
Hi everyone,

below is a report of an issue I came across with

**Environment**
- 1.36.33-bullseye1
- Zoneminder was installed from latest instructions (https://wiki.zoneminder.com/Debian_11_Bullseye_with_Zoneminder_1.36.x)
- Linux zoneminder-bullseye 5.10.0-3-amd64 #1 SMP Debian 5.10.13-1 (2021-02-06) x86_64 GNU/Linux

**Describe the bug**
I can't view mp4 live feed for url "/zm/cgi-bin/nph-zms?mode=mpeg&format=mp4&monitor=1&maxfps=30&bitrate=1000000"

**To Reproduce**
Currently I test it in command line with command 

`ffmpeg -i "http://{server IP}/zm/cgi-bin/nph-zms?mode=mpeg&format=mp4&monitor=1&maxfps=30&bitrate=1000000" -vcodec copy output.mp4`

**Expected behavior**
Live stream should be opened

**Debug Logs**
```

"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,INF,"Backtrace complete, please execute the following command for more information: addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x55594dd76e28 0x7f14b6839140 0x55594dd5318e 0x55594dd53909 0x7f14b682dea7 0x7f14b479ca2f",zm_signal.cpp,108
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f14b479ca2f]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7ea7) [0x7f14b682dea7]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x5e909) [0x55594dd53909]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x5e18e) [0x55594dd5318e]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x13140) [0x7f14b6839140]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x81e28) [0x55594dd76e28]",zm_signal.cpp,101
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Signal address is (nil), from 0x55594dd5318e",zm_signal.cpp,80
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89381,ERR,"Got signal 11 (Segmentation fault), crashing",zm_signal.cpp,50
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,ERR,"Could not open codec. Error code -542398533 ""Generic error in an external library""",zm_mpeg.cpp,296
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,WAR,"profile is optional; x264 defaults to high",zm_ffmpeg.cpp,69
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,WAR,"speed presets are listed in x264 --help",zm_ffmpeg.cpp,69
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,WAR,"preset usage: -vpre <speed> -vpre <profile>",zm_ffmpeg.cpp,69
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,WAR,"use an encoding preset (e.g. -vpre medium)",zm_ffmpeg.cpp,69
"4/16/23, 12:41:33 AM GMT+2",zms_m1,,89379,WAR,"broken ffmpeg default settings detected",zm_ffmpeg.cpp,69

```

VVD, 2023-04-16:
Probably I got the same issue.
Work jpeg stream only.
Mpeg streams doesn't work with different combinations of values for options MPEG_LIVE_FORMAT, FFMPEG_FORMATS and other.

Zoneminder 1.36.33, FreeBSD 13.2, ffmpeg 4.4.4, apache 2.4.57, php 8.1.18.

connortechnology, 2023-04-16:
That code hasn't been used in years, and I think it only ever supported mpeg1...  definitely needs an update.

nejcmedved, 2023-04-17:
could I help in any way?

if you can give me some guidelines how to approach the issue I am willing to help. I can develop/debug software in c++, python and HTML/Js.

Would really be willing to dig in the bug, some initial guidance would help a lot.

Best, Nejc

SteveGilvarry, 2024-09-13:
Fix streaming for 1.38?

connortechnology, 2024-09-13:
We need a better way to stream video.  Probably websocket based with an appropriate player in js.  There are examples on github that we could likely use.  I think this has to go on 1.39.

I would say that for 1.38 we need to remove the no longer functional options.

closed 2025-12-29 by SteveGilvarry