ZoneMinder/zoneminder #3695: Dlandon branded unraid docker image has malicious activity

baudneo, 2023-04-28:
A user on Reddit has seemingly found a dlandon zoneminder unraid image that has suspicious crypto activity in the db logs. IF YOU UAVE RECENTLY DOWNLOADED ANY DLANDON IMAGES, CHECK YOUR INSTALLS!

It is unknown what the attack is at this point, but please, check your installs.

dlandon, 2023-04-28:
I pushed that new image today to be sure the docker container is not compromised!  It's not nefarious activity!

I am responding on discorsd and am trying to figure out what is happening to one user.  So far one it's only one user that has reported this issue.  It is not widespread.

You are be overreacting on this issue.  I run the Zoneminder docker myself and have never seen this issue.  There's no proof that it is a compromised docker container.

Check this out: https://www.imperva.com/blog/hundreds-of-vulnerable-docker-hosts-exploited-by-cryptocurrency-miners/

Please take it down a notch until we understand where this is coming from.

baudneo, 2023-04-28:
If you read all the posts ive made I make it clear we don't know what's going on. I'd rather be safe than sorry and I'll wear whatever mess that entails.

dlandon, 2023-04-28:
> If you read all the posts ive made I make it clear we don't know what's going on.

Not really.  That's not what you said here.

> New info is suggesting that it is the official dlandon unraid docker image for ZoneMinder. The attackers pushed a new image shortly after the Reddit post was made to possibly hide the nefarious image.

That user reporting the issue needs to delete the Zoneminder data and do a fresh install to confirm if it's a docker issue.

mperf, 2023-04-29:
I'm the reddit User, I can confirm that right after a clean install I get always the same logs. Here is the steps I made for a clean install so You can confirm I did nothing wrong.

- deleted the docker container from the docker unraid tab (with the 'also delete image' option ticked)
- used the plug-in 'CA cleanup appdata' for deleting all appdata files
- finally check if both image and appdata files are deleted with a terminal
- re-installed from the apps section in unraid

dlandon, 2023-04-29:
Please post your diagnostics on the Unraid forum so we can have a look.  There are a lot of very sharp users there can can help us determine the source of this issue.  I'll be able to help more if I can have access to your diagnostics.

I'm concerned that one of your other docker containers or something you added is causing this.  The best bet is to boot in safe mode and install the Zoneminder docker container by itself.  The link I showed you in the reddit post suggests that docker containers can access other docker containers and cause problems.  That might be the case here.

dlandon, 2023-04-29:
User has reported on Reddit that the issue seems to be with the Brave Browser.  I think you can close this as it's a non issue.

baudneo, 2023-04-30:
Going to let it simmer for a bit, but I will say from the way it's looking that I am sorry and it does look like a non issue. 

I would rather err on the side of caution and look like a fool than have harm come to any of our user base.

baudneo, 2023-04-30:
@mperf if you could chime in here and confirm whatever findings or conclusions.

mperf, 2023-05-01:
Yes, sorry for late response, I figured out what the problem is. These error logs comes from a web extension called "phantom", It is a web3 wallet so there's no malware or such thing. I'm sorry for the noise that made but It was really strage for me since I didn't know that the zm's log page logs browser warnings and errors. Thanks everyone for the help

baudneo, 2023-05-01:
Alright, I will close the issue as it seems to be resolved. Just glad it's not an issue! Sorry @dlandon but, like I said, I'd rather err on the side of caution

closed 2023-05-01 by baudneo