ZoneMinder/zoneminder #3701: Errors keep repeating after adding and then removing a monitor

Simpler1, 2023-05-10:
**Describe Your Environment**
- ZoneMinder v 1.37.40 from [zoneminder-base](https://github.com/zoneminder-containers/zoneminder-base) nightly
- Full name and version of OS: Linux Mint 20.3 Cinnamon
- Browser: Chrome Version 112.0.5615.165 (Official Build) (64-bit)

**Describe the bug**
Add a new monitor, then delete the monitor.
The log file keeps repeating the following errors:
```
2023/05/10-12:27:41 | zms_m3 | 308370 | ERR | Unable to connect to monitor id 3 for streaming | zm_stream.cpp | 59
2023/05/10-12:27:41 | zms_m3 | 308370 | ERR | Can't open memory map file /dev/shm/zm.mmap.3: No such file or directory | zm_monitor.cpp | 862
```
[image]

Even restarting ZoneMinder does not resolve this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a new monitor
2. Delete the monitor
3. View the Error Logs

**Expected behavior**
When a monitor is deleted, the corresponding processes associated to that monitor will also be deleted.

connortechnology, 2023-05-10:
I just added some code to sort of deal with this... in terms of sending an instructive jpeg. but if the browser asks for the stream of monitor 3, it is going to log an error.  Now... a page refresh should cause the browser to know that the monitor is deleted.  So which view is up in your browser trying to view monitor 3?

SteveGilvarry, 2024-09-13:
@Simpler1 Did it get sort of fixed?

Simpler1, 2024-09-19:
Can no longer reproduce, so it seems to be fixed.

closed 2024-09-19 by Simpler1