ZoneMinder/zoneminder #3702: Janus request rtsp url doesn't contain password and username

sophipl, 2023-05-11:
**Describe Your Environment**
Zoneminder v1.37.40
running on
```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.10
Release:        22.10
Codename:       kinetic

uname -a
Linux recorder 5.19.0-41-generic #42~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 18 17:40:00 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

**Describe the bug**
Janus doesn't work
```
[ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_connect_to_server:7455] Couldn't get DESCRIBE code: 401
[ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:4030] Error creating 'rtsp' stream...
```

**Debug Logs from janus**
```json

{"janus" : "message", "transaction" : "randomString", "body" : {"request" : "create", "admin_key" : "KEY_HERE", "type" : "rtsp", "rtsp_quirk" : true, "url" : "rtsp://10.235.1.32:554/h265Preview_01_main", "secret" : "PASSWORD_HERE", "pin" : "PIN_HERE", "rtsp_user" : "admin", "rtsp_pwd" : "PASSWORD_HERE", "id" : 1, "video" : true}}

```

I track the issue to zm_monitor_janus.cpp
shouldn't `rtsp_path` be updated with username and password?
https://github.com/ZoneMinder/zoneminder/blob/8676da6eadbf3c4449a150b1b259ff73ad957b34/src/zm_monitor_janus.cpp#L101
At very least this line has no effect, since before it, we have the same variable assignment
https://github.com/ZoneMinder/zoneminder/blob/8676da6eadbf3c4449a150b1b259ff73ad957b34/src/zm_monitor_janus.cpp#L96

connortechnology, 2023-05-11:
Yeah I would say you are totally correct...

My question is... how is any of this working for anybody?!

connortechnology, 2023-05-11:
I think we actually pass the auth in separate fields when talking to janus. I don't think we actually need to put it back into the url. I did a PR that does so, but not sure if needed.

connortechnology, 2023-05-11:
I have a camera that just gets 401's despite auth being correct and my PR doesn't fix it. I notice that you are trying to stream h265 which I think also will not work. Ithink webrtc is limited to h264, vp8, vp9

sophipl, 2023-05-11:
Ok, I see, that's why it doesn't work for me.
So no h265 for me :(
I was wondering if the password as separate field was used in Janus.

jp-bennett, 2023-05-11:
Janus takes the username and password as separate fields, [here](https://github.com/ZoneMinder/zoneminder/blob/8676da6eadbf3c4449a150b1b259ff73ad957b34/src/zm_monitor_janus.cpp#L202). 

This is probably the libcurl bug, something we've fought for a while. Is Janus installed as a system binary, and what version of libcurl is installed on your system?

jp-bennett, 2023-05-11:
> Yes, Janus is installed as system binary

This is a libcurl problem, then. https://github.com/meetecho/janus-gateway/issues/3038

We've had good success with running Janus in docker, to get a known good libcurl. https://hub.docker.com/r/baudneo/janus or https://hub.docker.com/r/canyan/janus-gateway

sophipl, 2023-05-11:
Well baudneo didn't work
[image]

canyan didn't very differently
[janus.log](https://github.com/ZoneMinder/zoneminder/files/11456189/janus.log)

I have no experience with Janus, so I might not configured something, in fact I have 3 days experience with IP camera recording software :)

sophipl, 2023-05-12:
Well I dug more into this, tcpdumped the container, and it seems that initially we get 401 because of attempt to connect without credentials.
But we hang somewhere in janus javascript call

[image]

[image]

Pending call to janus js code
[image]

jp-bennett, 2023-05-12:
@sophipl one thing that might catch you, is the version of janus.js that get's included in the ZM web interface. It needs to match the version/revision of Janus that is running.

sophipl, 2023-05-12:
Yes, I considered it and made that they match, but that doesn't help somehow
[image]

It just doesn't work

baudneo, 2023-06-01:
Latest master rbanch of curl fixes the RTSP auth issue. You need to compile the new curl, remove cxurrent curl and libcurl and link new libcurl in. You will need build tools first, after the build env is setup you can issue these commands on debain based systems ->

** PLEASE BE AWARE OF WHAT YOU ARE DOING **
`sudo dpkg --remove --force-depends` can do some damage, just because it worke don my machine doesnt mean its universal. If things go wrong, reinstall libcurl4 and curl `sudo apt install libcurl4 curl`

```bash
git clone  https://github.com/curl/curl.git
cd curl
autoreconf -fi
./configure --with-openssl
make
sudo dpkg --remove --force-depends libcurl4
sudo dpkg --remove --force-depends curl
sudo make install
sudo ldconfig
curl --version
```

sophipl, 2023-06-03:
Hi

I'm currently on vacations, in a week I will be back and try curl compilation.
Thanks for the warning, I broken and repaired so many times my Linux OS (mainly the GPU drivers and libc), that I'm familiar :)
But I think something else is wrong as well, even when I use the docker Janus it doesn't work, and it communicates with the cameras when I capture the packets from the container.
I wish I would had some error log in this case, that I can try to fix.
I just saw that Janus JS object has closed the connection, from what I remember it was Promise type object and I didn't see what triggered it.

jp-bennett, 2023-06-03:
> even when I use the docker Janus it doesn't work

I rebuilt my camera server this last week, and discovered that several of the Janus Docker images have grabbed the libcurl release with this bug, too.

baudneo, 2023-06-03:
Docker image that works for x86_64 -> baudneo/janus:testing

See https://GitHub.com/baudneo/janus-gateway-docker for docker compose file.

sophipl, 2023-06-11:
I'm afraid it didn't work
One of docker images I used before was baudneo.
I wish I had more info what's the problem
Janus logs are fine, and even before I shown in wireshark, that janus connects to camera and works.

[image]

baudneo, 2023-06-11:
What do your Janus debug logs say? 
Edit janus.jcfg and enable debug logging at level 7 to a log file. Restart Janus, make sure debug logging is working and then retry. The Javascript part seems to be working fine. Need to see 3hqts happening when Janus tried connecting to the camera.

baudneo, 2023-06-11:
Also, did you try forcing the profile level I'd?

sophipl, 2023-06-11:
Yes, I tried with, and without profile id
Attaching a log

[janus_7.log](https://github.com/ZoneMinder/zoneminder/files/11715738/janus_7.log)

Firefox has "more to say", and it gets 465 Error processing SDP
[image]

baudneo, 2023-06-11:
Mozilla is notorious for being picky, have you tried chrome?

sophipl, 2023-06-11:
Yes, by default, the previous console screenshot is from chrome

baudneo, 2023-06-11:
Janus does not support h265 due to licensing costs. There is an exception for Safari browser see [here](https://www.meetecho.com/blog/av1-h265-janus/). H264 is supported and I think AV1 works.

sophipl, 2023-06-11:
I see, I had never owned apple devices, so no safari for me.
Does ZoneMinder has licensing for H265?, or just the Linux libraries support it somehow, since I can MJPED it? :)

baudneo, 2023-06-11:
I am unsure about ZM and how they handle H265, I *think* FFmpeg is the one who bought licensing rights and ZM just uses their hooks. 
Janus is a totally separate org and software stack though. There is a [video interview](https://www.youtube.com/watch?v=vGyXenlhj-o) that @jp-bennett did with Lorenzo from Janus where he explains the issue and why Janus cannot support H265.

sophipl, 2023-06-11:
Well, I switched to H264 and it didn't work either.
But since I want to have H265 codec, Janus is not for me :(

baudneo, 2023-06-11:
Ah, well if you do want to try h264 I can try debugging with you.

sophipl, 2023-06-11:
If I could make preview on one stream (H264), recording on 4K (H265), and movement analysis on third stream or downscaled H265.
Currently I'm running on one stream, 4K H265, and downscale it (720p) for movement analysis and preview, and doing ffmpeg passthrough for recording (so it record 4K).
Cameras have 4K streams in H265 and H264, and 640x480 on H264 stream (maybe H265 as well).
Zoneminder doesn't seem to work correctly with multiple streams I noticed, or wasn't doing what I expect it to do so.

sophipl, 2023-06-11:
FYI, apparently chrome version since 107 supports H265, if OS supports it (some claims that only Chrome OS, some claim 11gen Intel CPU).
https://www.chromium.org/audio-video/#:~:text=codec%20and%20container%20support

And I think my chrome works with H265, because when I download event, it opens browser tab with in browser player, and when I save the file (ctrl+s), the OS tells me that the file is H.265.
I run Windows 11 on video preview machine.

jp-bennett, 2023-06-12:
> chrome version since 107 supports H265, if OS supports it

Oh interesting. I don't know if I have a device to try that with, even.

sophipl, 2023-06-12:
Janus confuses me greatly.
But since when I switch to H264 I get the same result.
I configured Janus the same way this video says, perhaps something is wrong here?

https://www.youtube.com/watch?v=otAVgAMSUB8 (is it your video? @jp-bennett)

My setup is a bit more complicated, since the camera interface is vlan interface, and the output interface is the same interface without vlan tagging, perhaps here we have problem with interface binding?

baudneo, 2023-06-12:
Can you post the debug logs for the h264 stream. We've found some quirks and their workarounds. Whatever is happening, it's when Janus connects to the cam to setup pulling the stream in.

closed 2023-06-13 by sophipl