ZoneMinder/zoneminder #3711: Montage with layout 2 wide flips to vertical after an alert/alarm

Simpler1, 2023-05-23:
**Describe Your Environment**
- Version of ZoneMinder: 1.37.40
- Installed using docker zoneminder-containers/zoneminder-base:nightly-1170
- Linux Mint 20.3 Cinnamon
- Chrome Version 112.0.5615.165 (Official Build) (64-bit)

**Describe the bug**
With 2 monitors defined, when viewing the Montage screen with the layout set to "2 Wide", as soon as an alert -> alarm -> alert is triggered, the display flickers (as one of the views gets larger and smaller by a few pixel) and then the monitors become vertical instead of horizontal.  This is very irritating.

**To Reproduce**
Steps to reproduce the behavior:
1. Define two monitors
2. Open the Montage view
3. Switch the layout to "2 Wide"
4. Trigger an alert/alarm/alert

**Expected behavior**
The layout should remain as 2 wide after a an alert is triggered

connortechnology, 2023-05-23:
I've certainly seen this many times over the years, but not for a while now.  There is a css conflict on the .alert css class.  

It isn't happening here.  Are you on dark css or classic or something?

Simpler1, 2023-05-23:
I'm running out of box Skin = Classic, CSS = Base.  I've pulled my hair out yesterday trying to track down which css setting is the cause, but couldn't figure it out.

Simpler1, 2023-05-24:
I found that this issue occurs in Chrome, but doesn't occur with Firefox.

Simpler1, 2023-05-24:
Ah Ha!

```
.alert {
    position: relative;
    padding: .75rem 1.25rem;
    margin-bottom: 1rem;
    border: 1px solid ￼transparent;
    border-radius: .25rem
}
```
is in
`/cache/css_bootstrap.min-base-1684544183.css`

The padding css line should be removed.

Now the question is where to change this in the code.  :)

PR #3713 (2023-05-24): (fix): Montage 2 wide is stable on Chrome during alert
bootstrap has in internal css class called .alert that interferes with ZoneMinder's .alert class.

This fix overrides the bootstrap class and sets padding to 0.

This fixes #3711
closed 2023-05-24 by connortechnology