ZoneMinder/zoneminder #3717: Storage: 0 Events Using Null (when events exist)

Alcatraz077, 2023-05-31:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]: v1.36.33 
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]: Ubuntu VM, installed via PPA from Isaac Connor, ie: ```sudo add-apt-repository ppa:iconnor/zoneminder-1.36 && sudo apt update -y && sudo apt install zoneminder ```
- Full name and version of OS: Ubuntu 22.04.2LTS (Ubuntu Jammy)
- Browser name and version (if this is an issue with the web interface): N/A

**Describe the bug**
Storage table shows ```Events: 0 using null```

**To Reproduce**
Steps to reproduce the behavior:
1. Install Zoneminder on Ubuntu using recommended PPA
2. Add camera
3. Have event occur
4. Check Console and verify event occurred
5. Check Storage tab, note that storage is used, but showing ``` Events: 0 using null```

**Expected behavior**
Events to show within the table, as this also caused an issue where the AutoPurge didn't fire, because it didn't see any of the events as actually existing (even though I can play/download events).

**Debug Logs** (I ran this at Log Level 2, then triggered an event, this is the output):
```
Date/Time,Component,Server,PID,Level,Message,File,Line
"5/31/23, 5:48:46 PM CDT",web_php,,1944,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:48:46 PM CDT",web_php,,1944,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:48:45 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280600 - Capturing at 20.00 fps, capturing bandwidth 718782bytes/sec Analysing at 20.20 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:40 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280500 - Capturing at 20.00 fps, capturing bandwidth 854506bytes/sec Analysing at 20.40 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:36 PM CDT",web_php,,1850,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:48:36 PM CDT",web_php,,1850,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:48:35 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280400 - Capturing at 20.00 fps, capturing bandwidth 735019bytes/sec Analysing at 20.40 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:32 PM CDT",web_php,,1850,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:48:32 PM CDT",web_php,,1850,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:48:30 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280300 - Capturing at 20.00 fps, capturing bandwidth 844692bytes/sec Analysing at 19.00 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:25 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280200 - Capturing at 20.00 fps, capturing bandwidth 717819bytes/sec Analysing at 20.60 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:20 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280100 - Capturing at 20.00 fps, capturing bandwidth 856001bytes/sec Analysing at 19.60 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:15 PM CDT",zmc_m1,,1778,INF,"Living Room: 1280000 - Capturing at 19.94 fps, capturing bandwidth 734212bytes/sec Analysing at 19.94 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:10 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279900 - Capturing at 20.01 fps, capturing bandwidth 844820bytes/sec Analysing at 19.81 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:08 PM CDT",web_php,,16545,DBG,"View: request Request: status Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:48:08 PM CDT",web_php,,16545,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:48:05 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279800 - Capturing at 20.01 fps, capturing bandwidth 721576bytes/sec Analysing at 20.41 fps",zm_monitor.cpp,1680
"5/31/23, 5:48:00 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279700 - Capturing at 19.99 fps, capturing bandwidth 851497bytes/sec Analysing at 19.59 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:57 PM CDT",web_php,,1851,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:57 PM CDT",web_php,,1851,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:55 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279600 - Capturing at 20.00 fps, capturing bandwidth 734953bytes/sec Analysing at 20.60 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:50 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279500 - Capturing at 20.00 fps, capturing bandwidth 843668bytes/sec Analysing at 19.40 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:45 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279400 - Capturing at 20.00 fps, capturing bandwidth 724123bytes/sec Analysing at 20.20 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:40 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279300 - Capturing at 19.97 fps, capturing bandwidth 849622bytes/sec Analysing at 19.57 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:39 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:39 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:35 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279200 - Capturing at 19.99 fps, capturing bandwidth 735990bytes/sec Analysing at 20.39 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:34 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:34 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:32 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:32 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:30 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279100 - Capturing at 19.99 fps, capturing bandwidth 844336bytes/sec Analysing at 19.40 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:29 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:29 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:25 PM CDT",zmc_m1,,1778,INF,"Living Room: 1279000 - Capturing at 20.00 fps, capturing bandwidth 718544bytes/sec Analysing at 20.40 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:24 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:24 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
"5/31/23, 5:47:20 PM CDT",zmc_m1,,1778,INF,"Living Room: 1278900 - Capturing at 20.01 fps, capturing bandwidth 854564bytes/sec Analysing at 19.81 fps",zm_monitor.cpp,1680
"5/31/23, 5:47:19 PM CDT",web_php,,1848,DBG,"View: request Request: log Action:  User: admin",/usr/share/zoneminder/www/index.php,212
"5/31/23, 5:47:19 PM CDT",web_php,,1848,DBG,"Setting cookie parameters to Array(    [lifetime] => 3600    [path] => /    [domain] =>     [secure] =>     [httponly] => 1    [samesite] => Strict)",/usr/share/zoneminder/www/includes/session.php,44
[... cut ...]
"5/31/23, 5:44:00 PM CDT",zmc_m1,,1778,INF,"Living Room: 1274900 - Capturing at 20.00 fps, capturing bandwidth 849807bytes/sec Analysing at 19.60 fps",zm_monitor.cpp,1680
"5/31/23, 5:43:55 PM CDT",zmc_m1,,1778,INF,"Living Room: 1274800 - Capturing at 19.96 fps, capturing bandwidth 695752bytes/sec Analysing at 20.55 fps",zm_monitor.cpp,1680
```

Zoneminder itself is running fine, I just ran into this and was confused as to why. 

Thanks for your help, and please let me know if I need to submit more verbose logs.

Alcatraz077, 2023-06-01:
I can confirm that worked, thanks so much!

closed 2023-06-05 by connortechnology

jeremysherriff, 2024-09-29:
Sorry to resurrect an old issue, but is there a way to update the database entries to reflect the new id=1 storage location from the current unassigned/null value?  I have tried executing a filter for `move` and also `copy` and then `update used disk space` but have not been successful in having the storage usage reflect the moved/copied events.  Is my only option to wait for data to age out, or move to an alternate location and then back?

connortechnology, 2024-09-30:
You can talk directly to the database.  mysql -u zmuser -p zm 
UPDATE Events SET StorageId=1 WHERE StorageId IS NULL OR StorageId=0;