ZoneMinder/zoneminder #3728: Zoneminder not respecting the WEB_EVENTS_PER_PAGE variable any more

DagNygren, 2023-06-19:
**Describe Your Environment**
- Version of ZoneMinder: v1.36.33
- How you installed ZoneMinder :  RPMFusion
- Full name and version of OS: Fedora 38
- Browser name and version Firefox 114.0 

**Describe the bug**

The variable WEB_EVENTS_PER_PAGE worked earlier but now Zoneminder completely ignores it and always displays 10 events per page Also clicking on one item and returning with the back button will not always return to the previous page, but to the first page of the events.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options -> Web and set WEB_EVENTS_PER_PAGE to say 60 and save the setting
2. Click on Console
3. Select the event list for a camera
4. Check how many rows you see on the page

**Expected behavior**
Expect to see 60 events in the list

- Note: This is regression. It did work some time ago. Cannot say when though.

covidbishop, 2023-08-25:
I have the same problem. There's no way to set permanently the events number per page to other value than 10 with ZM ver.1.36.33.
It was working fine on v.1.36.26.

OkWhatIsAvailable, 2024-05-19:
Well, It's been a year since this was opened.  Same problem with v.1.36.33.  Wonder if anyone is looking at it.

connortechnology, 2024-05-21:
Please note that the current events list saves the value in a cookie. If a value has already been set, then the default will not be applied.

commit 101bf62a (2024-05-21): Set default value for rows per page using WEB_EVENTS_PER_PAGE.  Fixes #3728
commit d4c0203a (2024-05-21): Set default value for rows per page using WEB_EVENTS_PER_PAGE.  Fixes #3728
closed 2024-05-21 by connortechnology