ZoneMinder/zoneminder #3730: Debian Bookorm - can't update/install ZoneMinder v1.37.x via zmrepo

jzahraoui, 2023-06-20:
Hello :)
this not a bug but simply a kind of feature request.
want to be able to get the benefit of the 1.37 version on my installation.
I based on the previous ticket : https://github.com/ZoneMinder/zoneminder/issues/3665

**Describe Your Environment**- 
- Full name and version of OS : debian bookworm stock

**Describe the bug**
I can see that no builds are presents on the repo : https://zmrepo.zoneminder.com/debian/master/bookworm/

tried to install unbutu releases with this apt configuration :
deb https://ppa.launchpadcontent.net/iconnor/zoneminder-master/ubuntu kinetic main

but dependency do not resolve.

It si possible to install it from sources ? and what will be the procedure ?

connortechnology, 2023-06-27:
You should be able to now. We have packages available for amd64

jzahraoui, 2023-06-27:
thank you it work like a charm.

closed 2023-06-28 by connortechnology

mtk11, 2023-07-12:
@connortechnology How about packages for armhf ?