ZoneMinder/zoneminder #3731: ZM uses hard coded paths (like /bin/rm and uname)

ctheune, 2023-06-23:
**Describe Your Environment**
- v1.36.33
- NixOS
- NixOS 22.05

**Describe the bug**

NixOS relies heavily on injecting proper PATH variables to assemble package dependencies. The standard FHS structure does not exist so things like /bin/rm are not around. Applications and libraries on Unix should almost NEVER use absolute path names for those utilities due to the kind of forward-portability issues like this.

The `uname` issue causes spurious warnings in the log.

The "/bin/rm" issue causes events not being deleted properly, specifically audit not doing its job.

**To Reproduce**

Run it on nixos ... ;)

**Expected behavior**

I expect external dependencies to be used by allowing the regular PATH lookups to happen. 

**Debug Logs**
```

Jun 23 19:24:34 fc42 zmaudit.pl[836441]: Filesystem event 4787 at /var/lib/zoneminder/events/3/2023-05-16/4787 does not exist in database and is 3043609 seconds old, deleting
Jun 23 19:24:34 fc42 zmaudit.pl[836604]: Can't exec "/bin/rm": No such file or directory at /nix/store/xcf3g22rfp3axx2f5g9p0zhlq95gywla-zoneminder-1.36.33/lib/perl5/site_perl/5.34.1/ZoneMinder/General.pm line 59.

```

There's at least two occurences for "/bin/rm". And here's a full search for all "/bin", "/sbin" references. As NixOS includes `/bin/sh` and `/usr/bin/env` I have dropped those. In case that you want an easy fix you can prefix all those with `/usr/bin/env` if you don't want to deep dive into PATH lookups.

```
lib/perl5/site_perl/5.34.1/ZoneMinder/Event.pm:472:        my $command = "/bin/rm -rf $storage_path/$event_path";
lib/perl5/site_perl/5.34.1/ZoneMinder/Storage.pm:128:    my $command = "/bin/rm -rf $storage_path/$path 2>&1";
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:246:      `/usr/bin/zmdc.pl stop $process -m $$monitor{Id}`;
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:248:      `/usr/bin/zmdc.pl stop zma -m $$monitor{Id}`;
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:249:      `/usr/bin/zmdc.pl stop zmc -m $$monitor{Id}`;
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:254:      `/usr/bin/zmdc.pl start $process -m $$monitor{Id}`;
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:256:      `/usr/bin/zmdc.pl start zmc -m $$monitor{Id}`;
lib/perl5/site_perl/5.34.1/ZoneMinder/Monitor.pm:257:      `/usr/bin/zmdc.pl start zma -m $$monitor{Id}`;
share/zoneminder/db/zm_update-1.20.1.sql:135:insert into Controls values (1,'pelco-d','Local','/usr/local/bin/zmcontrol-pelco-d.pl',1,1,0,1,1,0,0,1,NULL,NULL,NULL,NULL,1,0,3,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,20,1,1,1,1,0,0,0,1,1,NULL,NULL,NULL,NULL,1,0,63,1,254,1,NULL,NULL,NULL,NULL,1,0,63,1,254,0,0);
share/zoneminder/db/zm_update-1.20.1.sql:136:insert into Controls values (2,'visca','Local','/usr/local/bin/zmcontrol-visca.pl',1,1,0,1,0,0,0,1,0,16384,10,4000,1,1,6,1,1,1,0,1,0,1536,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,3,1,1,1,1,0,1,1,0,1,-15578,15578,100,10000,1,1,50,1,254,1,-7789,7789,100,5000,1,1,50,1,254,0,0);
share/zoneminder/db/zm_update-1.20.1.sql:137:insert into Controls values (3,'KX-HCM10','Remote','/usr/local/bin/zmcontrol-kx-hcm10.pl',0,0,0,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,8,1,1,1,0,1,0,0,1,1,NULL,NULL,NULL,NULL,0,NULL,NULL,0,NULL,1,NULL,NULL,NULL,NULL,0,NULL,NULL,0,NULL,0,0);
share/zoneminder/db/zm_update-1.20.1.sql:138:insert into Controls values (4,'pelco-d-full','Local','/usr/local/bin/zmcontrol-pelco-d.pl',1,1,0,1,1,0,0,1,NULL,NULL,NULL,NULL,1,0,3,1,1,0,0,1,NULL,NULL,NULL,NULL,0,NULL,NULL,1,1,0,1,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,1,0,1,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,20,1,1,1,1,0,0,0,1,1,NULL,NULL,NULL,NULL,1,0,63,1,254,1,NULL,NULL,NULL,NULL,1,0,63,1,254,0,0);
share/zoneminder/db/zm_update-1.21.0.sql:21:insert into Controls values (0,'PELCO-P','Local','/usr/local/bin/zmcontrol-pelco-p.pl',1,1,0,1,1,0,0,1,NULL,NULL,NULL,NULL,1,0,3,1,1,0,0,1,NULL,NULL,NULL,NULL,0,NULL,NULL,1,1,0,1,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,1,0,1,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,20,1,1,1,1,0,0,0,1,1,NULL,NULL,NULL,NULL,1,0,63,1,254,1,NULL,NULL,NULL,NULL,1,0,63,1,254,0,0);
share/zoneminder/db/zm_update-1.21.0.sql:22:INSERT INTO Controls VALUES (0,'Axis API v2','Remote','/usr/local/bin/zmcontrol-axis-v2.pl',0,0,0,1,0,0,1,0,0,9999,10,2500,0,NULL,NULL,1,1,0,1,0,0,9999,10,2500,0,NULL,NULL,1,1,0,1,0,0,9999,10,2500,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,0,0,0,0,0,NULL,NULL,NULL,NULL,0,NULL,NULL,1,12,1,1,1,1,1,0,1,0,1,-360,360,1,90,0,NULL,NULL,0,NULL,1,-360,360,1,90,0,NULL,NULL,0,NULL,0,0);
share/zoneminder/www/api/app/Console/cake.php:1:#!/usr/bin/php -q
share/zoneminder/www/api/lib/Cake/Console/cake.php:1:#!/usr/bin/php -q
share/zoneminder/www/api/lib/Cake/Console/Templates/skel/Console/cake.php:1:#!/usr/bin/php -q
share/zoneminder/www/ajax/shutdown.php:40:  #exec('sudo -n /bin/systemctl poweroff -i 2>&1', $data['output'], $data['rc']);
share/zoneminder/www/ajax/shutdown.php:42:  #ZM\Debug("Shutdown output " . shell_exec('/bin/systemctl poweroff -i 2>&1'));
share/zoneminder/www/ajax/shutdown.php:46:  #exec('sudo -n /bin/systemctl reboot -i 2>&1', $data['output'], $data['rc']);
```

PS: the last 3 lines seem scary even though they seem to be commented out ...

Simpler1, 2023-06-26:
@ctheune What do you propose for the replacements that would be Linux distribution agnostic?
```
/bin/rm  => ??
/usr/bin/php => ??
/bin/systemctl => ??
```

ctheune, 2023-07-11:
Just noticed that #3670  is another instance of this

Jhadevenz, 2023-07-17:
hahahaha

ctheune, 2023-09-27:
@connortechnology Anything I can help with here? I'm currently working around this by running a manual timer that cleans things up ...

connortechnology, 2023-09-27:
I was having trouble deciding which way to go.  I think I like adding a config option because it gives the user the ability to specify an executable that isn't the one on the path.  Also gives the ability to just use path... So I'll add 

ZM_PATH_RM=rm and ZM_PATH_UNAME=uname then packagers can use a full path if they want.

ctheune, 2023-09-27:
Sure, that's something I can work with. This does/should basically be repeated for all external binary dependencies. I'm happy to support either way, it just might be a lot more work for you in the long run.

I do understand the sentiment though: other distributions have a harder time constructing a fitting PATH environment.

Thanks!

connortechnology, 2023-09-29:
Possibly address by db390a4861f2a558981fb23f398ec84b464e6015. Comments?

ctheune, 2023-09-30:
Looks great! I can absolutely work with that!

closed 2023-09-30 by ctheune

Jhadevenz, 2023-12-23:
Hillo

Jhadevenz, 2023-12-23:
We have a fore sale game clash of clans