ZoneMinder/zoneminder #3739: Log is "orange" but stuck "Loading, please wait ..."

ctheune, 2023-07-11:
**Describe Your Environment**
- v1.36.33
- nixos module
- NixOS 23.05
- Safari

**Describe the bug**

The web UI shows the log menu as "orange". Unfortunately, clicking on it doesn't give me a log at the moment but is stuck showing "Loading, please wait ...".

I double-checked that the browser inspector doesn't show any errors and the systemd logs don't show errors either. Not sure where to keep looking ...

However, it does seem to work in Chrome!

**To Reproduce**

Use Safari to open the Log in the Web UI

**Expected behavior**

I expected to get a log ;)

**Debug Logs**

Didn't find anything. :(

connortechnology, 2023-07-14:
Pretty weird.  I just tried in safari on my macbook, works fine.  

Normally when we see something like this it is because there are too many records and php is hitting it's memory limit, which by default is 128Mb.  Bumping that up in php.ini might help.  Doesn't explain why it works in chrome though. Also, there should be warnings in the js console about it.  Best place to check is in apache logs (or cgi logs or whatever is appropriate.).  Those error don't tend to make it to our logging functions and happen at a higher level.

SteveGilvarry, 2024-09-13:
OP never responded with any logs, closing as stale

closed 2024-09-13 by SteveGilvarry

connortechnology, 2024-09-13:
I had a thought today along the lines of... can we check the size of php's used ram against the mem limit setting and bail gracefully if we are about to run out? Maybe.