ZoneMinder/zoneminder #3757: Zmc process for Anpviz camera consumes all system memory

neillbell, 2023-08-23:
**Describe Your Environment**

- Version of ZoneMinder: 1.37.42 (1.37.42~20230818165840-focal_amd64)
- How you installed ZoneMinder: ppa:iconnor/zoneminder-master
- Full name and version of OS: Ubuntu 20.04.6 LTS

**If the issue concerns a camera**
- Make and Model: Anpviz IPK-3108P262W-4S (YMF52_STARIR_NM223_AF)
- frame rate: 20fps
- frame interval: 20
- resolution: 1920x1080
- ZoneMinder Source Type: RTSP/TCP

**Describe the bug**
The zmc process for the affected camera consumes all memory and swap available until crashing, presumably with an out-of-memory error. No error messages are produced by the process.

This behavior began with the installation of the zoneminder_1.37.42\~20230810110346-focal_amd64 package and continues to the latest version. Reverting to zoneminder_1.37.42\~20230803113254-focal_amd64 causes the zmc processes to return to normal behavior.

Changes to frame rate, frame interval, video compression and encoding profile do not affect the problem.

Cameras from Hikvision, Vivotech, Uniview and the R-Series sold by Nelly's Security are not affected.

**Debug Logs**
```
==> zmc_m32.log <==
08/23/23 07:47:04.187946 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 500 - Capturing at 19.94 fps, capturing bandwidth 313731bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:09.180511 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 600 - Capturing at 20.03 fps, capturing bandwidth 323823bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:14.188890 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 700 - Capturing at 19.97 fps, capturing bandwidth 308548bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:19.180506 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 800 - Capturing at 20.03 fps, capturing bandwidth 322551bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:24.192936 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 900 - Capturing at 19.95 fps, capturing bandwidth 307705bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:29.187225 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 1000 - Capturing at 20.02 fps, capturing bandwidth 322745bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:34.187727 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 1100 - Capturing at 20.00 fps, capturing bandwidth 309754bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:39.192062 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 1200 - Capturing at 19.98 fps, capturing bandwidth 313907bytes/sec Analysing at 0.00 fps]
08/23/23 07:47:44.191878 zmc_m32[909853].INF-zm_monitor.cpp/1790 [Street: 1300 - Capturing at 20.00 fps, capturing bandwidth 313222bytes/sec Analysing at 0.00 fps]

==> zmdc.log <==
08/23/2023 07:47:50.232200 zmdc[905861].INF [ZMServer:717] ['zmc -m 32' crashed, signal 8]
08/23/2023 07:48:00.245205 zmdc[905861].INF [ZMServer:412] [Starting pending process, zmc -m 32]
08/23/2023 07:48:00.263924 zmdc[905861].INF [ZMServer:412] ['zmc -m 32' starting at 23/08/23 07:48:00, pid = 910866]
08/23/2023 07:48:00.270696 zmdc[910866].INF [ZMServer:412] ['zmc -m 32' started at 23/08/23 07:48:00]

```

neillbell, 2023-08-23:
I just tried setting "Maximum Image Buffer Size" to 130 and that seems to prevent the problem. That camera does not allow the keyframe interval to be set below the frame rate. Is using MaxImageBuffers good enough for production use in this case?

connortechnology, 2023-08-23:
I think there is an actual bug here, which I am fixing.

I do recommend setting MaxImageBuffers to something. Can be as high as you are comfortable with. You are probably finding lots of warnings in your logs. Also analysis and recording may not be happening.

connortechnology, 2023-12-08:
As I do not have your camera, I have no idea if it is resolved. Please do an update and let us know.

neillbell, 2023-12-09:
I upgraded and it looks good. Thanks Isaac!

neillbell, 2023-12-09:
Forgot to close with comment

closed 2023-12-09 by neillbell