ZoneMinder/zoneminder #3759: Remote/RTSP is broken: Failed to bind RTCP server

rrygl, 2023-08-26:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] Debian 1.36.33+dfsg1-1
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] Debian stock
- Full name and version of OS: Debian
- Browser name and version (if this is an issue with the web interface): Irrelevant

**Describe the bug**
Remote/RTSP is broken with error log: "Failed to bind RTCP server"
I think the bug was introduced in 71edb9d830696a618e9f8a5911c748bbf4356f92. Before this commit RtpCtrlThread needs to call .start() but this commit changed it to start without it.
Then in zm_rtsp.cpp there are ctrl thread that doesn't want to start:
```
      // These never actually run
      RtpDataThread rtpDataThread( *this, *source );
      RtpCtrlThread rtpCtrlThread( *this, *source );
```
Now it starts and fail.

**To Reproduce**
Simply use Remote/RTSP.

**Expected behavior**
It should work

**Debug Logs**
```
08/26/23 01:44:53.536731 zmc_m11[9752].DB2-zm_rtsp.cpp/574 [RTSP Seq is 44366]
08/26/23 01:44:53.536737 zmc_m11[9752].DB2-zm_rtsp.cpp/575 [RTSP Rtptime is 2099095912]
08/26/23 01:44:53.536760 zmc_m11[9752].DB3-zm_rtp_source.cpp/59 [RTP CName = zm-11@server]
08/26/23 01:44:53.536767 zmc_m11[9752].DB3-zm_rtp_source.cpp/92 [Initialising sequence]
08/26/23 01:44:53.536969 zmc_m11[9753].DB2-zm_rtp_data.cpp/65 [Starting data thread 2131313403 on port 0]
08/26/23 01:44:53.536990 zmc_m11[9754].DB2-zm_rtp_ctrl.cpp/252 [Starting control thread 7f0942fb on port 1]
08/26/23 01:44:53.537061 zmc_m11[9754].DB1-zm_comms.cpp/543 [bind(): Trying '0.0.0.0', family '2', proto '17']
08/26/23 01:44:53.537072 zmc_m11[9753].DB1-zm_comms.cpp/543 [bind(): Trying '0.0.0.0', family '2', proto '17']
08/26/23 01:44:53.537096 zmc_m11[9753].DB3-zm_rtp_data.cpp/79 [Bound to :0]
08/26/23 01:44:53.537114 zmc_m11[9754].ERR-zm_comms.cpp/559 [bind(), Could not bind]
08/26/23 01:44:53.537124 zmc_m11[9754].FAT-zm_rtp_ctrl.cpp/264 [Failed to bind RTCP server]
```

connortechnology, 2023-08-28:
I'm not so sure... it looks to me like the port isn't configured and it isn't getting a hostname for localhost... What do you have in Options->Network->MIN_RTP_PORT ? Default is 40200.  You might try changing this...

rrygl, 2023-08-28:
MIN_RTP_PORT is still default 40200.

The error log shows that it tried to bind port **0** and **1** (comes from 0+1). Binding 0 is fine because it picks a random one instead. Binding 1 failed.
This simply means that the port number variables were only initial default values.

These two threads were not supposed to run at all (the comment says so). But yet they run and tried to bind port 0 and 1. That's the problem.

romale, 2023-09-05:
I've this issue too

glorang, 2024-02-14:
I have the same issue on Debian 11, zoneminder-1.36.33-bullseye1 from the zoneminder APT repo.

This is what I got from the debug output:

```
02/14/24 22:07:32.611078 zmc_m3[132786].DB2-zm_rtsp.cpp/474 [Got RTSP session BBD57073, timeout 60 secs]
02/14/24 22:07:32.611090 zmc_m3[132786].DB2-zm_rtsp.cpp/479 [Got RTSP transport RTP/AVP/TCP;unicast;interleaved=0-1;ssrc=001FF16F;mode="PLAY"]
02/14/24 22:07:32.611107 zmc_m3[132786].DB2-zm_rtsp.cpp/517 [RTSP Method is RTP/RTSP]
02/14/24 22:07:32.611116 zmc_m3[132786].DB2-zm_rtsp.cpp/518 [RTSP Distribution is unicast]
02/14/24 22:07:32.611124 zmc_m3[132786].DB2-zm_rtsp.cpp/519 [RTSP SSRC is 1ff16f]
02/14/24 22:07:32.611133 zmc_m3[132786].DB2-zm_rtsp.cpp/520 [RTSP Local Host is ]
02/14/24 22:07:32.611142 zmc_m3[132786].DB2-zm_rtsp.cpp/521 [RTSP Local Ports are 0/0]
02/14/24 22:07:32.611150 zmc_m3[132786].DB2-zm_rtsp.cpp/522 [RTSP Remote Ports are 0/0]
02/14/24 22:07:32.611159 zmc_m3[132786].DB2-zm_rtsp.cpp/523 [RTSP Remote Channels are 0/1]
02/14/24 22:07:32.611175 zmc_m3[132786].DB2-zm_rtsp.cpp/43 [Sending RTSP message: PLAY rtsp://cam3.domain.com:80/axis-media/media.amp?videocodec=h264&resolution=1280x960 RTSP/1.0^M
```
I dug a bit in the source (I'm not an expert!) but it seems that a new RtpSource object is created here in zm_rtsp.cpp:

```
  switch( mMethod ) {
    case RTP_RTSP :
    case RTP_RTSP_HTTP :
    {
      RtpSource *source = new RtpSource( mId, "", remoteChannels[0], mHost, remoteChannels[0], ssrc, seq, rtpClock, rtpTime, codecId );
    }
  }
```
Where `remoteChannels[0]` (3rd parameter) is passed further down to fill the values of `localPortBase` / `mLocalPortChans` which are used by `getLocalCtrlPort()`

But looking at the source (again in zm_rtsp.cpp):

```
  } else if (StartsWith(parts[i], "interleaved=") ) {
      method = "RTP/RTSP";
      StringVector subparts = Split(parts[i], "=");
      StringVector channels = Split(subparts[1], "-");
      remoteChannels[0] = strtol( channels[0].c_str(), nullptr, 10 );
      remoteChannels[1] = strtol( channels[1].c_str(), nullptr, 10 );
```

it seems to just parse the value from `interleaved=0-1` in the RTSP transport response and pass that as port number to the bind calls later on.

`MIN_RTP_PORT` is only used in `requestPorts()` but that is only called for type `RTP_UNICAST` it seems.

Hope this makes any sense and helps you track down the issue.

connortechnology, 2024-09-13:
Yeah remote should stick around but deprecated for rtsp.  

I feel like this should be fixed though.

commit 0af44191 (2024-09-13): If an invalid port is specified, don't actually start the rtp threads.  They don't get used in RTP/RTSP. Fixes #3759
commit 52550651 (2024-09-13): If an invalid port is specified, don't actually start the rtp threads.  They don't get used in RTP/RTSP. Fixes #3759
closed 2024-09-13 by connortechnology