ZoneMinder/zoneminder #3762: no serializer registered for SOAP version

romale, 2023-09-04:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
- ZM v1.36.33 
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
```
cat zmrepo-centos.repo 
[zmrepo]
name=Zoneminder and its dependencies for Enterprise Linux $releasever - $basearch
baseurl=http://zmrepo.connortechnology.com/el/$releasever/$basearch
        http://67.53.78.182/el/$releasever/$basearch
        http://zmrepo.mehvc.com/el/$releasever/$basearch
#mirrorlist=http://zmrepo.mehvc.com/el/$releasever/$basearch
enabled=1
gpgcheck=1
# zmrepo uses packages from other repos
# Import all required gpg keys
gpgkey=file:///etc/pki/rpm-gpg/RPM-GPG-KEY-EPEL-$releasever-zmrepo
       file:///etc/pki/rpm-gpg/RPM-GPG-KEY-rpmfusion-free-el-$releasever-zmrepo
       file:///etc/pki/rpm-gpg/RPM-GPG-KEY-zmrepo

```
- Full name and version of OS
```
cat /etc/redhat-release 
CentOS Linux release 7.9.2009 (Core)

```
**Describe the bug**

zmonvif-probe.pl returns  error

```
rpm -qf /usr/bin/zmonvif-probe.pl
zoneminder-common-1.36.33-3.el7.x86_64

/usr/bin/zmonvif-probe.pl -v probe http://192.168.88.245:8899/onvif/device_service
Probing for SOAP http://192.168.88.245:8899/onvif/device_service
no serializer registered for SOAP version http://192.168.88.245:8899/onvif/device_service at /usr/share/perl5/vendor_perl/SOAP/WSDL/Factory/Serializer.pm line 22.

/usr/bin/zmonvif-probe.pl -v probe http://192.168.88.245/onvif/device_service
Probing for SOAP http://192.168.88.245/onvif/device_service
no serializer registered for SOAP version http://192.168.88.245/onvif/device_service at /usr/share/perl5/vendor_perl/SOAP/WSDL/Factory/Serializer.pm line 22.
```

Curl returns some data

```
IP=192.168.88.245
curl -s -X POST http://$IP:8899/onvif/device_service\
>  -H 'Content-Type: application/soap+xml; charset=utf-8'\
>  -H 'SOAPAction: "http://www.onvif.org/ver20/media/wsdl/GetProfiles"'\
>  -d '
> <soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdl="http://www.onvif.org/ver20/media/wsdl">
>   <soap:Header/>
>   <soap:Body>
>     <wsdl:GetProfiles/>
>    </soap:Body>
> </soap:Envelope>
> ' | xmllint --format -
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xs="http://www.w3.org/2000/10/XMLSchema" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:ns1="http://www.w3.org/2005/05/xmlmime" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:ns7="http://docs.oasis-open.org/wsrf/r-2" xmlns:ns2="http://docs.oasis-open.org/wsrf/bf-2" xmlns:dndl="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dnrd="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:ns10="http://www.onvif.org/ver10/replay/wsdl" xmlns:ns11="http://www.onvif.org/ver10/search/wsdl" xmlns:ns13="http://www.onvif.org/ver20/analytics/wsdl/RuleEngineBinding" xmlns:ns14="http://www.onvif.org/ver20/analytics/wsdl/AnalyticsEngineBinding" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:ns15="http://www.onvif.org/ver10/events/wsdl/PullPointSubscriptionBinding" xmlns:ns16="http://www.onvif.org/ver10/events/wsdl/EventBinding" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:ns17="http://www.onvif.org/ver10/events/wsdl/SubscriptionManagerBinding" xmlns:ns18="http://www.onvif.org/ver10/events/wsdl/NotificationProducerBinding" xmlns:ns19="http://www.onvif.org/ver10/events/wsdl/NotificationConsumerBinding" xmlns:ns20="http://www.onvif.org/ver10/events/wsdl/PullPointBinding" xmlns:ns21="http://www.onvif.org/ver10/events/wsdl/CreatePullPointBinding" xmlns:ns22="http://www.onvif.org/ver10/events/wsdl/PausableSubscriptionManagerBinding" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:ns3="http://www.onvif.org/ver10/analyticsdevice/wsdl" xmlns:ns4="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:ns5="http://www.onvif.org/ver10/display/wsdl" xmlns:ns8="http://www.onvif.org/ver10/receiver/wsdl" xmlns:ns9="http://www.onvif.org/ver10/recording/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:trt2="http://www.onvif.org/ver20/media/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tnsn="http://www.eventextension.com/2011/event/topics">
  <SOAP-ENV:Header/>
  <SOAP-ENV:Body>
    <trt2:GetProfilesResponse>
      <trt2:Profiles fixed="true" token="000">
        <trt2:Name>Profile_000</trt2:Name>
        <trt2:Configurations>
          <trt2:VideoSource token="000">
            <tt:Name>VideoS_000</tt:Name>
            <tt:UseCount>3</tt:UseCount>
[... 283 lines cut ...]
                  <tt:Min>-1</tt:Min>
                  <tt:Max>1</tt:Max>
                </tt:XRange>
                <tt:YRange>
                  <tt:Min>-1</tt:Min>
                  <tt:Max>1</tt:Max>
                </tt:YRange>
              </tt:Range>
            </tt:PanTiltLimits>
            <tt:ZoomLimits>
              <tt:Range>
                <tt:URI>http://www.onvif.org/ver10/tptz/ZoomSpaces/PositionGenericSpace</tt:URI>
                <tt:XRange>
                  <tt:Min>-1</tt:Min>
                  <tt:Max>1</tt:Max>
                </tt:XRange>
              </tt:Range>
            </tt:ZoomLimits>
          </trt2:PTZ>
        </trt2:Configurations>
      </trt2:Profiles>
    </trt2:GetProfilesResponse>
  </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

connortechnology, 2023-09-05:
You didn't specify which version to use. Command should have been/usr/bin/zmonvif-probe.pl -v probe http://192.168.88.245:8899/onvif/device_service

romale, 2023-09-05:
Hi.
Thanks for your reply.
I just use cli format from command help

```
zmonvif-probe.pl 
Usage: /bin/zmonvif-probe.pl [-v] probe <soap versions> <network interface>
       /bin/zmonvif-probe.pl [-v] <command> <device URI> <soap version> <user> <password>
  Commands are:
    probe     - scan for devices on the local network and list them
    profiles  - print the device's supported stream configurations
    metadata  - print some of the device's configuration settings
    move      - move the device (only ptz cameras)
  Common parameters:
    -v        - increase verbosity
  Device access parameters (for all commands but 'probe'):
    device URL    - the ONVIF Device service URL
    soap versions - SOAP versions (1.1 or 1.2 or 1.1,1.2)
    user          - username of a user with access to the device
    password      - password for the user
```

Your cli return other error
```

zmonvif-probe.pl -v probe 1.2 http://192.168.88.245:8899/onvif/device_service
Probing for SOAP 1.2
Setting net interface for WSDiscovery::TransportUDP=SCALAR(0x286cdd0) to http://192.168.88.245:8899/onvif/device_service
unknown or unconfigured interace http://192.168.88.245:8899/onvif/device_service at /usr/share/perl5/vendor_perl/WSDiscovery/TransportUDP.pm line 78.
```

romale, 2023-09-05:
I tried with soap version, with username, without and without user password (my camera have no password). The error is same "no serializer registered for SOAP version"

```
zmonvif-probe.pl -v probe http://192.168.88.245:8899/onvif/device_service 1.1,1.2
Probing for SOAP http://192.168.88.245:8899/onvif/device_service
Setting net interface for WSDiscovery::TransportUDP=SCALAR(0x16b2dd0) to 1.1,1.2
no serializer registered for SOAP version http://192.168.88.245:8899/onvif/device_service at /usr/share/perl5/vendor_perl/SOAP/WSDL/Factory/Serializer.pm line 22.
```

romale, 2023-09-05:
"zmonvif-probe.pl -v profiles ..." worked fine, i think
```
zmonvif-probe.pl -v profiles http://192.168.88.245:8899/onvif/device_service 1.2
Have results from GetServices
Got device service http://192.168.88.245:8899/onvif/device_service
Got media service http://192.168.88.245:8899/onvif/Media
Got events service http://192.168.88.245:8899/onvif/Events
Got imaging service http://192.168.88.245:8899/onvif/Imaging
Have results from GetCapabilities: <GetCapabilitiesResponse xmlns="http://www.onvif.org/ver10/device/wsdl"><Capabilities><Analytics xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/analytics_service</XAddr><RuleSupport>true</RuleSupport><AnalyticsModuleSupport>true</AnalyticsModuleSupport></Analytics><Device xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/device_service</XAddr><Network></Network><System></System><IO><InputConnectors>0</InputConnectors><RelayOutputs>0</RelayOutputs><Extension><Auxiliary>false</Auxiliary><AuxiliaryCommands>nothing</AuxiliaryCommands><Extension></Extension></Extension></IO><Security></Security></Device><Events xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/Events</XAddr><WSSubscriptionPolicySupport>true</WSSubscriptionPolicySupport><WSPullPointSupport>true</WSPullPointSupport><WSPausableSubscriptionManagerInterfaceSupport>false</WSPausableSubscriptionManagerInterfaceSupport></Events><Imaging xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/imaging</XAddr></Imaging><Media xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/Media</XAddr><StreamingCapabilities><RTPMulticast>false</RTPMulticast><RTP_TCP>true</RTP_TCP><RTP_RTSP_TCP>true</RTP_RTSP_TCP></StreamingCapabilities><Extension><ProfileCapabilities><MaximumNumberOfProfiles>2</MaximumNumberOfProfiles></ProfileCapabilities></Extension></Media><PTZ xmlns="http://www.onvif.org/ver10/schema"><XAddr>http://192.168.88.245:8899/onvif/PTZ</XAddr></PTZ><Extension xmlns="http://www.onvif.org/ver10/schema"><Extensions></Extensions></Extension></Capabilities></GetCapabilitiesResponse>
Received message:
<?xml version="1.0"?>
<GetProfilesResponse xmlns="http://www.onvif.org/ver10/media/wsdl">
  <Profiles token="000" fixed="true">
    <Name xmlns="http://www.onvif.org/ver10/schema">Profile_000</Name>
    <VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema">
      <Name>VideoS_000</Name>
      <UseCount>3</UseCount>
...
```

connortechnology, 2023-09-06:
Right I screwed up with cutting and pasting.
probe does not take a URI.  The profiles command does. The detect cameras we do 
/usr/bin/zmonvif-probe.pl -v probe 1.1,1.2
To scan an actual camera we do
/usr/bin/zmonvif-probe.pl -v profiles

The profiles command is obviously working.

romale, 2023-09-08:
If I understand correctly, a camera somewhere on the Internet (behind the router) cannot be interrogated using probe?
The task is simply something like this:
add a camera via onvif to use motion detection using the camera (and moving camera by PTZ), and not ZM, which heavily loads the system on CPU and memory. And use ZM to record the moment of movement, which ZM should learn about from the camera (can it do this?). I didn’t really find anything about this in the ZM documentation. If this is in the documentation, please share the link if it’s not difficult for you.

SteveGilvarry, 2024-09-13:
Closing as stale. I think you are asking about onvif event detection

closed 2024-09-13 by SteveGilvarry