ZoneMinder/zoneminder #3766: Cameras show offline when they are not

PX03AFK, 2023-09-21:
**Describe Your Environment**
- Version of ZoneMinder 1.37.4 - just updated
- How you installed ZoneMinder from source
- Full name and version of OS - OpenSuse Leap 15.4
- Browser name and version (if this is an issue with the web interface) - Firefox 117.0.1

**If the issue concerns a camera**
3 x Foscam FI18918W; 30 fps; 640 x 480; HTTP
2 x Foscam FI8904W; 40 fps; 640 x 480; HTTP
1 x Foscam FI9912P; 25 fps; 640 x 480; TCP

On the ZM console one or more of the cameras will show offline and then, when the console refreshes, one or more or all of them will not be offline.  If a camera is marked offline I can still click on the Id or Name and get a view so the camera is actually there and working.

This didn't happen on version 1.37.3 which I had been using previously.

I switched debugging on and could refresh the console and when I can see that the state of a camera has changed the two debug logs that appear for the refresh don't have any other log in between and there is nothing in the main log file.

connortechnology, 2023-09-21:
1.37.4 is OLD.  You should be at 1.37.44.

PX03AFK, 2023-09-21:
My apologies, this is 1.37.44.  I obviously didn't press hard enough and even though I checked the report for accuracy I missed this

connortechnology, 2023-09-21:
Okay then I sorta know what the issue is.  I added code to check the age of the monitor status. Basically if it is over a minute old, it is disregarded.  So why are your monitor status entries so old?

PX03AFK, 2023-09-21:
What is it that is getting monitor status?  Some are ethernet connected so there should be no delay.  Can you advise me on what to look at to determine why they appear to be taking a long time?

Thanks for helping.

PX03AFK, 2023-09-22:
I also have a problem that when I try to view events with replay set to gapless events, only one event plays and I have to click forward for the next.  Could this be associated with this problem or is this something else?

PX03AFK, 2023-09-28:
I've been trying to find out where the UpdatedOn field in the Monitor_Status gets updated but I don't seem to be able to find the relevant piece of code.  Looking directly at the database just after restarting ZM UpdatedOn gets updated quite frequently but gradually, over time, some cameras don't get updated for around 10 minutes.  
I can make the console status look OK by changing the two places in console.php that check for 60 seconds into 600 seconds but that only tidies up the console view, not the underlying problem,

connortechnology, 2023-09-29:
Line 1778 of zm_monitor.cpp in UpdateFPS.  If you had fps_report_interval (Monitor->Misc) set high, then that might explain this...

PX03AFK, 2023-09-29:
I've never changed fps_report_interval and I find it set at 1000, which is what the current user guide suggests.  Most of my cameras are showing a frame rate of 1.83 fps so that would be the reason why this problem is showing up now.  Are there any other ramifications of changing fps_report_interval to a lower value.  Also, if the update is controlled by this setting, why doesn't the check in console.php use that value instead of the fixed value when the two things seem to be intertwined.

connortechnology, 2023-09-29:
Um, which user guide?  Default is 100. Agree that value should have been taken into account in console.

closed 2023-09-29 by connortechnology

PX03AFK, 2023-09-29:
I was looking in this guide: https://sgzoneminder.readthedocs.io/en/latest/.  As I've never changed this value on any of my cameras I can only assume that the default was 1000 when I originally set them up.  But that was years ago so I guess the default has changed.

connortechnology, 2023-09-29:
Yeah it likely has.  Probably around the 1.32 series.  Docs have now been updated.

mjnewton, 2024-08-19:
Did this fix definitely work as intended? Under v1.37.63 I keep seeing cameras appearing as Offline in the console if the FPS Report Interval is set high (e.g. 1000) despite the camera streams fully functioning. If I lower the interval to 100 the console gives a persistent clean sheet, albeit at the expense of my log file being heavily clobbered.

PX03AFK, 2024-08-20:
It did get rid of my problem and I haven't noticed any detrimental affect on the log file.

commit 10cf26fa (2023-09-29): Take FPSReportInterval into account when determining if a monitor is dead. Fixes #3766