ZoneMinder/zoneminder #3767: Onvif probe fail to detect profiles

allotherarenotavaliable, 2023-09-23:
**Describe Your Environment**
- Version of ZoneMinder 1.36.33
- How you installed ZoneMinder PPA
- Full name and version of OS ubuntu 22.04.3 LTS
- Browser name and version not related to browser

**If the issue concerns a camera**
- ZYsecurity
- frame rate 12
- 2592x1944 and 704x576
- ZoneMinder Source Type:

**Describe the bug**
This is camera problem and you need same model (or some other model with same problem) in order to reproduce bug.
Anyway, this can be fixed in onvif error handling.

From main window choose "Add" and in monitor configuration window click on onvif.
- In first step onvif is detecting camera.
- After entering username/password and "next" there are no profiles.

I have tested camera behavior (new in onvif, not perl programmer) and have detected :
GetProfiles is working without problem
GetStreamUri crash because of broken response
(I am guessing "next" is reading both profiles and GetStreamUri)

The problem is because character "&" is not properly encoded in response and that onvif crash because xml is not valid.

I have intercepted GetStreamUriResponse after GetStreamUri and broken part is in Uri

<tt:Uri>rtsp://10.10.10.10:554/user=admin_password=xxxxx=0_stream=0&onvif=0.sdp?real_stream</tt:Uri>

From my point of view solution would be to catch xml parse error, check raw output for malformed responses and try to parse response again. something like ... Replace("&", "&amp;");

(Yes, I understand this is camera wrong response, not soap bug but looks like many camera models have same problem and this would be nice improvement to handle error and fix this error)

I am new in zoneminder and not linux programmer but have a lot experience in IT so fell free to contact me for more details if this is not clear.

Camera is identified from broadcast with two lines for two soap version
I2P3C-model@<1.1.1.1> using version 1.[1-2]

Magical-Lin, 2023-10-10:
same OS ubuntu 22.04.3 LTS, my camera onvif port is 8090, but i cant not check my camera at onvif network searching. so can we add the funtion that custom onvif IP and port ?

allotherarenotavaliable, 2023-10-10:
> same OS ubuntu 22.04.3 LTS, my camera onvif port is 8090, but i cant not check my camera at onvif network searching. so can we add the funtion that custom onvif IP and port ?

Discovery (probe) is UDP multicast. I guess your camera is not onvif if this can not be find. Alternative is that you are not on same sub network or your router is not forwarding those packets

Magical-Lin, 2023-10-22:
> > same OS ubuntu 22.04.3 LTS, my camera onvif port is 8090, but i cant not check my camera at onvif network searching. so can we add the funtion that custom onvif IP and port ?
> 
> Discovery (probe) is UDP broadcast. I guess your camera is not onvif if this is not found. Alternative is that you are not on same sub network or your router is not forwarding those packets

but i can use ONVIF Device Manager to find it ,and can custom ip:port to connect it via onvier that android app.
[image]

connortechnology, 2023-10-24:
Can we please get the output from zmonvif-probe.pl -v profiles? 

I don't see why a & in a url would be invalid... is it not utf8? Is that what you are saying? Technically we could try to sanitise the response before soap parsing... but this feels very wrong.

Magical-Lin, 2023-10-25:
> Discovery (probe) is UDP broadcast. I guess your camera is not onvif if this is not found. Alternative is that you are not on same sub network or your router is not forwarding those packets

Currently, zoneminder  does not support custom IP to automatic setting function, so i use odm2.2.2 to find the rtsp url that fill in Source Path, it work.
[image]

By the way， on Android apps, if searching funtion is not work , we  can custom onvif url, just simply fill in the 'host address，port ,user and password' to automatically complete the settings, including PTZ. it is userfull in NAT traversal,  cross segment access.
Ah... at first i thought zoneminder could be like this.

SteveGilvarry, 2024-09-13:
Original issue appeared to be camera not returning valid XML from the SOAP request, problem being where in some library we use for onvif is the control to allow invalid XML responses, and parse them to resolve the issue? Going to close this out as non-onvif compliant camera unless someone comes back with a different opinion and we can reopen.

closed 2024-09-13 by SteveGilvarry