ZoneMinder/zoneminder #3768: Thunderbird and iPhone email don't display images inline

akgrant43, 2023-09-24:
**Describe Your Environment**
- ZoneMinder 1.37.45
- How you installed ZoneMinder: PPA
- Ubuntu 22.04.3

**Describe the bug**
Images, such as added with the `%EI1% macro are added as attachments instead of inline.

**To Reproduce**
1. Add the `%EI1%` macro to the alarm email filter and produce an alarm.  E.g.:

```
ZM alarm detected

Monitor: %MN%
Time: %ET%

First alarm image: %EI1%

Highest score image: %EIMA%
```

**Expected behavior**
The images are displayed inline (not as attachments).

I'll be adding a PR soon(ish).

Thanks,
Alistair

SteveGilvarry, 2024-09-13:
@akgrant43 Did you ever PR this?

akgrant43, 2024-09-13:
Yes, https://github.com/ZoneMinder/zoneminder/pull/3769 , but...
it was closed without merging as inline images only work in html format, and I was using plain text.

closed 2024-09-13 by akgrant43