ZoneMinder/zoneminder #3773: Warning from zmfilter without camera or event id

PX03AFK, 2023-09-29:
Version of ZoneMinder 1.37.44 - just updated
    How you installed ZoneMinder from source
    Full name and version of OS - OpenSuse Leap 15.4
    Browser name and version (if this is an issue with the web interface) - Firefox 117.0.1

If the issue concerns a camera
3 x Foscam FI18918W; 30 fps; 640 x 480; HTTP
2 x Foscam FI8904W; 40 fps; 640 x 480; HTTP
1 x Foscam FI9912P; 25 fps; 640 x 480; TCP

I am seeing log messages about zmfilter not being able to delete events but it doesn't show a camera number or event id.  The only filter I have turned on is PurgeWhenFull and I haven't changed it from the original.  

**To Reproduce**
Allow PurgeWhenFull filter to run

**Expected behavior**
Error reports should include details of the monitor and the event.

**Debug Logs**
```
29/09/23 16:26:27.378340 zmfilter_1[32656].WAR [ZoneMinder::Event:380] [/opt/zoneminder/bin/zmfilter.pl Can't Delete event  from Monitor  StartDateTime:undef from main:342]

```

connortechnology, 2023-09-29:
Can we get a debug log? It's pretty wierd.  I think in order for this to happen... Filter->Execute would have to return something other than an event row... but looking at the code I don't see how that could happen.

PX03AFK, 2023-09-29:
I am having a problem getting a debug log.  I've set enable LOG_DEBUG in Options and also put _zmfilter in LOG_DEBUG_TARGET and set LOG_LEVEL_SYSLOG and LOG_LEVEL_FILE to Debug but I am still not getting any debug data.  What am I missing?

connortechnology, 2023-09-29:
Did you restart the filter?  

You can all do sudo kill -SIGUSR1 pid_of_filter which will increase the debug log level.

PX03AFK, 2023-09-29:
[zm_debug.log.6777.txt](https://github.com/ZoneMinder/zoneminder/files/12769667/zm_debug.log.6777.txt)

Managed to get a debug log this time.

PX03AFK, 2023-09-29:
I have run the SQL against the database and the query is returning a record where everything is NULL.  However if I do a straight select with an ID of NULL nothing comes back.
Is there a setting I have wrong within the database?

connortechnology, 2023-09-29:
On newer mysql, we get an error ERROR 1140 (42000): In aggregated query without GROUP BY, expression #1 of SELECT list contains nonaggregated column 'zm.E.Id'; this is incompatible with sql_mode=only_full_group_by. Because it is missing a GROUP BY E.Id.  

There were fixes for this since 1.37.44. Please update and check that this is still a problem.

connortechnology, 2023-09-29:
With debug on? That's weird.  You can always run it from command line to see if there is any uncaught output.  If debug is off though it shouldn't really log anything unless your drive is full.

PX03AFK, 2023-09-29:
Sorry, this was with debug off.  I've got used to seeing something that not having anything was very unusual, but good if that means nothing is going wrong.

connortechnology, 2023-10-24:
I think we have this one fixed. Closing.

closed 2023-10-24 by connortechnology