ZoneMinder/zoneminder #3776: Gapless/All events do not play automatically

PX03AFK, 2023-10-05:
Version of ZoneMinder 1.37.45 - recently updated
How you installed ZoneMinder from source
Full name and version of OS - OpenSuse Leap 15.4
Browser name and version (if this is an issue with the web interface) - Firefox 117.0.1

**Describe the bug**
I thought I had seen a report about this problem but I don't seem to be able to find it now. Apologies if this is a duplicate.

When viewing events setting replay to Gapless Events or All Events has no effect so I have to click on the Next button to move forward.  I am also seeing the speed reset to 1 and replay reset to None but at this moment I have not been able to determine when this happens

**To Reproduce**
1. On console click on a number in the Events column
2. On the list of events click on an event ID
3. The event will play with replay set to Gapless Events I would expect it to go to the next Event automatically, at least that is what happened with earlier versions.

**Expected behavior**
The next event should play without having to click Next

**Debug Logs**
I see nothing being logged about this and I did try with and without Log_Debug.

```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

PX03AFK, 2023-10-12:
I don't know if there is one problem or if there are two separate ones but something happens with the speed and Replay action.  I look at my events on a daily basis and whenever I put up the first event the speed has gone back to1 and Replay has been reset to None.  That is until I found the option WEB_L_DEFAULT_RATE which, from its description, may have something to do with the default speed of replay.  I changed this and sure enough when I open an event it plays faster.  However, even having set this value back to 100 when I view an event it now shows at 16x speed.

connortechnology, 2023-10-12:
FYI 16x is the max speed that the html video tag supports.  So an invalid DEFAULT_RATE will result in I dont know what behaviour.  

These settings are stored in cookies, so anything that clears cookies might cause this.  I'll have to do some testing.

PX03AFK, 2023-10-18:
Does anyone else have the problem of having to click the next button when viewing events?

connortechnology, 2023-10-19:
I have not yet been able to reproduce this issue. 

Is this using zms to playback or video.js? What is selected for Codec at top right?  Try different values. 

 Because I am aware of an issue where in some cases zms can't get to the end of the video....because  of a difference in timestamps between the db frames and the mp4 frames.  

It occurs to me after looking at the code, that the selected value for replaymode won't persist between logins... so I am adding it to possible values for the new User Preferences settings...

PX03AFK, 2023-10-19:
I am simply clicking on the first event in the event list for a particular monitor.  I've not specifically made any option changes (at least I am not aware of any I have made).  In an earlier version of 1.37 the playback would be at whatever speed I had selected for any previous playback and the Replay option was set to Gapless Events.  I've been working this way for years and the values persisted over logins, and system reboots.  Now, not only does the Gapless Events not persist but every so often the option for the number of rows visible on the event list goes back to 10 when I normally set it to ALL.

At present the Codec option on the view screen is Auto but I did try the other two on offer: MP4 and MJPEG, which didn't help.  I also tried All Events for the replay option but that doesn't help (as a matter of interest what is the difference between All Events and Gapless Events?).

I've just spotted something which may or may not have anything to do with this but I need to tell you and see if there may be a bug and therefore if a new report is necessary.  I've observed more than one instance where a playback is trying to find a file with a filename one more than the Frame count.  I had this at one point some time ago and at that time it looked as though the problem was when ZM was restarted and I could see errors being logged.  In this instance I can see no errors occurring when the original event is recorded.

connortechnology, 2023-10-24:
All Events will display a black screen and a timer until the next event happens.  Not sure who would ever want that. 

I would like to know if there are any javascript errors being logged.: Right click inspect, go to Console tab.

PX03AFK, 2023-10-24:
Sorry, should have thought to look at browser before.  Yes, there are two errors when I click on the Next.  Can you read this OK?

TypeError: globalThis.crypto.randomUUID is not a function
[inject.js:2648:54](moz-extension://12995599-c9c6-4309-add5-5faa5b97cc86/public/js/inject.js)
XHRGET
http://linuxdt/zoneminder/index.php?&command=99&connkey=786147&view=request&request=stream
[HTTP/1 500 Internal Server Error 125ms]

Request Failed: error, Internal Server Error [skins_classic_js_skin-base-1696011124.js:813:11](http://linuxdt/zoneminder/cache/skins_classic_js_skin-base-1696011124.js)
Ajax request failed.  No responseText.  jqxhr follows: [skins_classic_js_skin-base-1696011124.js:815:13](http://linuxdt/zoneminder/cache/skins_classic_js_skin-base-1696011124.js)
Object { readyState: 4, getResponseHeader: getResponseHeader(e), getAllResponseHeaders: getAllResponseHeaders(), setRequestHeader: setRequestHeader(e, t), overrideMimeType: overrideMimeType(e), statusCode: statusCode(e), abort: abort(e), state: state(), always: always(), catch: catch(e)
, … }
[skins_classic_js_skin-base-1696011124.js:816:13](http://linuxdt/zoneminder/cache/skins_classic_js_skin-base-1696011124.js)

​

connortechnology, 2023-10-25:
The first is not coming from ZM and I suspect may be from a browser plugin or malware.
The second says that one of our ajax requests, specifically the one that asks zms how the streaming is going failed.  You might find more information in apache logs or one of the zms_e logs.

So those don't shed a ton of light.

PX03AFK, 2023-10-25:
Appreciate you continuing to help investigate this.  

As a result of your prompt I've found the initial problem.  The extensions option wasn't in my apache so create_socket didn't work.  Having installed the extensions and turned them on that error has gone away but another has come up and I can't quite see what it means.  The error, as reported in the apache error log, is: Undefined index: default in /srv/www/htdocs/zoneminder/includes/Object.php on line 78.  Is that actually saying that there is an index missing from a table?  I'm afraid my ability to read php is not strong enough to work out why this is failing.

connortechnology, 2023-10-26:
I think you just need to update.  The recent Tags feature had an error which I have since fixed.

PX03AFK, 2023-10-31:
I've updated to 1.37.46 but still I can't get events to play automatically.  I've tried different browsers and different machines and see a mixture of different console messages but the end result is that the next event doesn't come up automatically.  Nothing is being logged.  Not sure what to do next.

connortechnology, 2023-11-01:
The zms crashing bit is a big clue. As I said before, either in zms logs or apache logs or web_php logs, there will be more information.  Look there and post what you see here.

connortechnology, 2023-11-01:
Okay, then I think I know what is going on.  We are playing back an mp4 using zms. There is a known problem where it sometimes can't actually get to the end of the file due to a mismatch between frame timestamps in the db and timestamps in the mp4.  I've been meaning to fix this for a while.  Perhaps I will get to it today.

PX03AFK, 2023-11-01:
That could be the case.  I've just had a look some events in the db and the filestore and I can where some files have times outside start and end times in the db.  
Could this also have anything to do with the issue of the number of events in the db being one more than in filestore?

connortechnology, 2023-11-01:
Please check latest.  Fixed a few things.  This commit should fix your problems: 2c1e573fd

connortechnology, 2023-11-01:
As to event counts being off... that can happen in a crash... you can run zmaudit.pl to fix it. event start/endtimes should always match the video.

PX03AFK, 2023-11-04:
Sorry, the No More Event data found message seems to have taken longer to appear than I expected.  It is there.

PX03AFK, 2023-11-05:
Still a little problem with the rate an type of replay.  I think you suggested it might be something to with cookies.  Whenever I first start replay after a break the speed is 1 and replay is set to none.  I can change those settings and they will remain for some time but I can't say just how long before they revert or if there is something else which resets the values.  It is clearly browser based as I can make the change in Firefox and switch over to Chrome and on Chrome it is still 1 and none.  Going back to Firefox the changed values are still there.  Are there some options playing into this?

PX03AFK, 2023-11-14:
OK, I think I can see the problem.  I decided to look at the cookies for replay and eventrate and their max_age is only one hour.  I have read something about browsers potentially limiting the maximum value for Expires.  I did try manipulating the value in the set_cookie function in session.php so that it was only trying to set one year but when the cookie changes it still has one hour.  Not sure if that is a function of the fact that the cookie already exists.

connortechnology, 2023-11-15:
There was some discussion about this a while back.  I figured an hour would be enough, but clearly that is not the case.  I will adjust the code to set it to a year.

PX03AFK, 2023-11-15:
Thanks Isaac.  I tend to look at my cameras on a daily basis but sometimes it goes to two or three days so 1 hour is a little short.  For me 1year would be better but that may not be the case for others.  Is there a way of having the value in the conf file or would that be too much trouble?  Just a thought.

connortechnology, 2023-11-20:
I think we have resolved this one.  cookie expiry was addressed in c23ee28fe6fd30a269e122c50eb109e5a86c0f51

closed 2023-11-20 by connortechnology

PX03AFK, 2023-11-21:
Thanks Isaac.  Just to confirm, the cookie change has fixed the issue.