ZoneMinder/zoneminder #3780: Read The Docs builds failing

efranzwa, 2023-11-01:
**Describe Your Environment**
This is my local environment for building ZoneMinder Read The Docs html 
- Version of ZoneMinder: development version 'master' 
- Version of python: 3.8.10
- Version of sphinx: 7.1.2
- Version of sphinx-rtd-theme: 1.3.0

**Describe the bug**
ZM Read the Docs build (on the Read the Docs site) is failing at this step:
python -m sphinx -T -E -W --keep-going -b html -d _build/doctrees -D language=en . $READTHEDOCS_OUTPUT/html

The error is :
ModuleNotFoundError: No module named 'sphinx_rtd_theme'
The build environment most likely did not install this module because our configuration files don't list it as a dependency.

**To Reproduce**
Steps to reproduce the behavior:
- This is a build failure on the Read the Docs site, you can see the fail logs here.
https://readthedocs.org/projects/zoneminder/builds/22402143/

**Expected behavior**
The build works fine on my local machine. The only difference is that I have installed sphinx-rtd-theme module in my python virtual environment.

** Suggested Solution **
- Follow the Read the Docs guide for setting up builds.  We need to modify the .readthedocs.yaml file and add a docs/requirements.txt file.
https://docs.readthedocs.io/en/stable/guides/reproducible-builds.html

- Add python dependencies to .readthedocs.yaml file

```
version: 2

build:
  os: "ubuntu-20.04"
  tools:
    python: "3.8"

sphinx:
  configuration: docs/conf.py
  fail_on_warning: true

python:
  install:
    - requirements: docs/requirements.txt
```

- Add a docs/requirements.txt file
```
sphinx == 7.1.2
sphinx-rtd-theme == 1.3.0
```

- Revert the commits that may not have not worked
- a43e715aa6f6de434a24cb7ddaf840b64381baad
- cfec797299b29fa6db86e70d5aad9801b053053f

**Debug Logs**
```
python -m sphinx -T -E -W --keep-going -b html -d _build/doctrees -D language=en . $READTHEDOCS_OUTPUT/html

Running Sphinx v7.1.2

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/config.py", line 356, in eval_config_file
    exec(code, namespace)  # NoQA: S102
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/checkouts/latest/docs/conf.py", line 17, in <module>
    import sphinx_rtd_theme
ModuleNotFoundError: No module named 'sphinx_rtd_theme'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/cmd/build.py", line 285, in build_main
    app = Sphinx(args.sourcedir, args.confdir, args.outputdir,
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/application.py", line 207, in __init__
    self.config = Config.read(self.confdir, confoverrides or {}, self.tags)
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/config.py", line 179, in read
    namespace = eval_config_file(filename, tags)
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/config.py", line 369, in eval_config_file
    raise ConfigError(msg % traceback.format_exc()) from exc
sphinx.errors.ConfigError: There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/config.py", line 356, in eval_config_file
    exec(code, namespace)  # NoQA: S102
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/checkouts/latest/docs/conf.py", line 17, in <module>
    import sphinx_rtd_theme
ModuleNotFoundError: No module named 'sphinx_rtd_theme'

Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/envs/latest/lib/python3.8/site-packages/sphinx/config.py", line 356, in eval_config_file
    exec(code, namespace)  # NoQA: S102
  File "/home/docs/checkouts/readthedocs.org/user_builds/zoneminder/checkouts/latest/docs/conf.py", line 17, in <module>
    import sphinx_rtd_theme
ModuleNotFoundError: No module named 'sphinx_rtd_theme'

```

efranzwa, 2023-11-01:
Isaac, I can try to implement the suggested fixes and revert the extra commits if you like (but probably faster if you try first ...).  Based on Read the Docs help info this should work.  Unfortunately I don't know how to test this prior to committing the changes as the real build occurs on the Read the Docs site.  I don't have a RTD account so not sure if it is possible to test prior to committing.

connortechnology, 2023-11-02:
I think we are good again.  Latest commits built successfully at readthedocs.io

closed 2023-11-02 by connortechnology

efranzwa, 2023-11-03:
Awesome! I checked the latest build at zoneminder.readthedocs.io and the **latest** version has applied the template correctly and also contains the updates pushed to the Documentation. Cheers!