ZoneMinder/zoneminder #3781: The github release asset for 1.36.33 cannot be built on its own.

Richard-Cranium, 2023-11-02:
By "release asset", I mean the source archives available at https://github.com/ZoneMinder/zoneminder/releases/tag/1.36.33

This isn't a runtime issue, this is a "cannot possibly build using" issue.  After downloading one of the release assets, un-zipping/tarring the source and changing to the resulting directory, the simple command "cmake ." fails with the earliest error message being...

**CMake Error at CMakeLists.txt:13 (message):
  The git submodules are not available.  Please run git submodule update
  --init --recursive**

Whatever you are using to create the release assets is not providing anything that can be used to build anything.  I'd prefer that you fix that so that those assets are actually useful, but I fear that you simply won't bother creating them any more.

Richard-Cranium, 2023-11-02:
Hi welcome bot!  Your template is tightly focused for runtime issues and there is no indication of how someone could be to report a non-runtime issue!

jp-bennett, 2023-11-02:
> Hi welcome bot!  Your template is tightly focused for runtime issues and there is no indication of how someone could report a build issue!

There are intelligent humans behind the curtain. I would tend to agree that the release tarball should have the git submodules populated.

SteveGilvarry, 2023-11-02:
Historically it was not possible to alter the releases and include the
submodule, probably a discussion in the issue history. But that is probably
not the case with GitHub actions. Thanks for the heads up.

connortechnology, 2023-11-02:
I wouldn't be opposed to doing away with the submodules.  They are a frequent source of annoyance.

Richard-Cranium, 2023-11-04:
> Historically it was not possible to alter the releases and include the submodule, probably a discussion in the issue history. But that is probably not the case with GitHub actions. Thanks for the heads up.

I don't expect you to fix the past; I simply hope that you'd fix the future or tell me that you cannot.

Richard-Cranium, 2023-11-04:
> > Hi welcome bot!  Your template is tightly focused for runtime issues and there is no indication of how someone could report a build issue!
> 
> There are intelligent humans behind the curtain. I would tend to agree that the release tarball should have the git submodules populated.

There are semi-intelligent humans (despite my avatar and nick) in front of the curtain as well.  It would have churlish of me to vent any annoyance to the welcome bot.

SteveGilvarry, 2023-11-04:
I do appreciate your nick, self proclaimed or earned title?

Richard-Cranium, 2023-11-04:
My nick here is self-proclaimed.  My gmail username is earned.  :D