ZoneMinder/zoneminder #3784: Monitor never stops capturing

Firstyear, 2023-11-13:
**Describe Your Environment**
- Version of ZoneMinder  - zoneminder-1.36.33-3.el8.x86_64
- How you installed ZoneMinder - docker built from https://github.com/ZoneMinder/zmdockerfiles/blob/master/release/el8/Dockerfile

- Full name and version of OS:  Linux a9a9d90b497c 5.14.21-150500.55.36-default #1 SMP PREEMPT_DYNAMIC Tue Oct 31 08:37:43 UTC 2023 (e7a2e23) x86_64 x86_64 x86_64 GNU/Linux

**If the issue concerns a camera**
- Make and Model: Unifi UVC G3 Flex
- frame rate: 25
- resolution: 1920x1080
- ZoneMinder Source Type: rtsp

**Describe the bug**
The camera is added as a monitor changing only the values:
* general > name
* function > monitor
* source > path rtsp:// ...
* source > resolution 1080p

When viewed as a monitor it works, but then the zmc process continues to run indefinitely and never ends, consuming 60% cpu constantly (2.1ghz xeon). This persists even if all references to the web page closes.

**Expected behavior**
zmc to stop after the monitor is closed, preventing the high cpu.

While the logs claim a lack of space in shm, this isn't correct:

```
# df -h
Filesystem                  Size  Used Avail Use% Mounted on
/dev/mapper/vg_system-root   58G   44G   14G  77% /
tmpfs                        64M     0   64M   0% /dev
tmpfs                        79G     0   79G   0% /sys/fs/cgroup
shm                         2.0G   24M  2.0G   2% /dev/shm
/dev/mapper/vg_system-root   58G   44G   14G  77% /etc/hosts
tmpfs                       2.0G  4.0K  2.0G   1% /run/httpd
nvme/srv/zm/log              20G  4.8G   16G  24% /var/log/zoneminder
nvme/srv/zm/db               40G   15G   26G  38% /var/lib/mysql
tmpfs                        79G     0   79G   0% /run/secrets/credentials.d
ssd/zm_data                  90G  6.9G   84G   8% /var/lib/zoneminder/events
tmpfs                        79G     0   79G   0% /proc/acpi
tmpfs                        79G     0   79G   0% /proc/scsi
tmpfs                        79G     0   79G   0% /sys/firmware

# ls -al /dev/shm/
total 24308
drwxrwxrwt 2 root   root         60 Nov 13 12:24 .
drwxr-xr-x 5 root   root        340 Nov 13 12:24 ..
-rw-r----- 1 apache apache 24888760 Nov 13 12:24 zm.mmap.1
```

It could be that zm isn't clearing the shm area properly at startup?

Firstyear, 2023-11-13:
**Debug Logs**
```

 * Starting MySQL database server service   ...done.
ERROR 1396 (HY000) at line 1: Operation CREATE USER failed for 'zmuser'@'localhost'
 * ZoneMinder dB already exists, skipping table creation.
 * Starting php-fpm web service   ...done.
 * Starting Apache http web server service   ...done.
 * Starting ZoneMinder video surveillance recorder
Database already at version 1.36.33, update skipped.

Freshening configuration in database
Migratings passwords, if any...
Loading config from DB 227 entries
Saving config to DB 227 entries
   ...done.
==> /var/log/zoneminder/zma_m1.log <==
09/06/23 15:10:02.815182 zma_m1[304].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
09/06/23 15:39:52.961082 zma_m1[306].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-deck]
09/07/23 08:52:13.000180 zma_m1[302].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-deck]
09/11/23 04:02:53.697350 zma_m1[10392].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
09/12/23 10:12:53.905325 zma_m1[304].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
09/20/23 18:11:20.921822 zma_m1[14329].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
09/25/23 18:10:11.560923 zma_m1[306].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
10/01/23 10:00:53.598454 zma_m1[23189].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
10/02/23 09:18:24.658166 zma_m1[26119].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.1, probably not enough space free: No such file or directory]
10/04/23 15:07:37.631000 zma_m1[5332].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-deck]

==> /var/log/zoneminder/zma_m3.log <==
08/15/23 16:17:26.828587 zma_m3[318].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-benny]
08/17/23 14:25:12.431240 zma_m3[3152].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
08/25/23 11:55:56.325183 zma_m3[14589].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-benny]
08/25/23 11:55:58.091681 zma_m3[14611].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
08/27/23 12:25:41.974008 zma_m3[312].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
09/12/23 09:58:45.450336 zma_m3[329].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
09/20/23 18:11:21.259607 zma_m3[14344].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
09/25/23 18:10:11.912713 zma_m3[322].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
10/01/23 10:00:53.990665 zma_m3[23207].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]
10/02/23 09:18:13.820402 zma_m3[26035].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.3, probably not enough space free: No such file or directory]

==> /var/log/zoneminder/zma_m4.log <==
08/25/23 11:55:56.620268 zma_m4[14604].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-side]
08/25/23 11:55:58.360557 zma_m4[14614].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.4, probably not enough space free: No such file or directory]
09/06/23 15:10:03.436197 zma_m4[336].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-side]
09/06/23 15:39:53.516469 zma_m4[338].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-side]
09/12/23 09:56:20.517795 zma_m4[328].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-side]
09/12/23 09:58:45.667260 zma_m4[345].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.4, probably not enough space free: No such file or directory]
09/20/23 18:11:31.870478 zma_m4[14414].ERR-zm_monitor.cpp/501 [Shared data not initialised by capture daemon for monitor bne1-side]
09/25/23 18:10:11.974496 zma_m4[338].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.4, probably not enough space free: No such file or directory]
10/01/23 10:01:04.617383 zma_m4[23273].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.4, probably not enough space free: No such file or directory]
10/02/23 09:18:14.253240 zma_m4[26050].ERR-zm_monitor.cpp/567 [Can't open memory map file /dev/shm/zm.mmap.4, probably not enough space free: No such file or directory]

==> /var/log/zoneminder/zma_m5.log <==
[... cut ...]
ZoneMinder Telemetry Agent starting at 23/11/13 12:24:27
Collecting data to send to ZoneMinder Telemetry server.

==> /var/log/zoneminder/zmstats.log <==
11/13/23 12:24:28.207256 zmstats[464].INF [main:40] [Stats Daemon starting in 30 seconds]

==> /var/log/zoneminder/zmwatch.log <==
11/13/23 12:24:27.794261 zmwatch[456].INF [main:79] [Watchdog starting, pausing for 30 seconds]

==> /var/log/zoneminder/zmtelemetry.log <==
11/13/23 12:24:29.380059 zmtelemetry[460].WAR [main:196] [Telemetry server returned HTTP POST error code: 301]

```

connortechnology, 2023-11-13:
This is by design.  In 1.37 we have a mode called Capturing:  OnDemand where it only captures when someone is viewing it.

Firstyear, 2023-11-14:
Hmmm, that's a bit counter intuitive because the way "monitor" reads it should only capture when it's being ... well, monitored. 

I guess I'll wait for 1.37 then.

closed 2023-11-20 by connortechnology