ZoneMinder/zoneminder #3798: Build failure for Ubuntu: `version.txt: No such file or directory`

LukeSavefrogs, 2023-12-28:
Following [this guide](https://zoneminder.readthedocs.io/en/stable/installationguide/ubuntu.html#harder-way-build-package-from-source) I tried building the deb package from source using the `stable` branch, but it fails at line 183:

connortechnology, 2023-12-29:
You need to use do_debian_package from the stable branch in order to build the stable branch.

LukeSavefrogs, 2023-12-29:
> You need to use do_debian_package from the stable branch in order to build the stable branch.

As i told in the issue i already did that. The stable release gets the latest STABLE tag and compiles from there, which leads to the bug explained up here

closed 2023-12-29 by connortechnology

LukeSavefrogs, 2023-12-29:
Ty @connortechnology for pushing the fix

commit f224bf72 (2023-12-29): Use either version or version.txt. Fixes #3798
commit b24955fb (2024-05-19): Use either version or version.txt. Fixes #3798