ZoneMinder/zoneminder #3801: [No Audio] Besder h265 passthrough - wont save with audio

porterdcarnegie, 2024-01-04:
Zoneminder v1.37.48 git master 2024-01-04.  My machine: Gentoo, ffmpeg 4.4.4-r8
Camera: Besder (cheap-o), h265 HEVC **only** 20fps@1080p, chipset: XM540.

Issue: When enabling storing audio in Recording option while using passthrough, no mp4 is saved at all, the log says:
`AVCodecContext was allocated for aac, but pcm_alaw passed to avcodec_open2()`

Manually it works, ffmpeg can save a mp4 with playable audio by: `ffmpeg -i "rtsp://foo.bar -c:v copy -c:a aac out.mp4`.

mpv reports audio format as:  pcm_alaw (PCM A-law / G.711 A-law), Channels: 1, Format: s16, Sample Rate: 8000Hz, Bitrate: 64kbps

connortechnology, 2024-01-04:
I believe this was fixed last night in 6d0f634bfc9dffc548fade14ba0d538e2e80f83f

porterdcarnegie, 2024-01-04:
Yes, video is now audible! I was a couple of commits behind.

closed 2024-01-04 by porterdcarnegie