ZoneMinder/zoneminder #3802: Onvif unreliable off event causes endless recordings

porterdcarnegie, 2024-01-04:
Zoneminder v1.37.48 git master 2024-01-04.
Camera: Besder (cheap-o), h265 HEVC only 20fps@1080p, chipset: XM540.

Issue: Camera is sporadically not sending `onvif off` events or seems like `onvif on/off` events are inverted.
This has the effect that alert state will be left on after motion finished and New Event will keep recording forever.

I worked around this by setting the two occurrences of Event_Poller_Closes_Event to false in zm_monitor.cpp.
Maybe there's no real solution other than a toggle option to disable this since it could be the camera that is malfunctioning.

SteveGilvarry, 2024-09-13:
Assume something with onvif code ignoring max event length. I don't have a camera working with onvif events at the moment to test this.

connortechnology, 2024-09-13:
Thing is, the monitor will stay in alarmed mode, so it will just create another event when it hits max event length.  We may need an onvif specific max alarmed time setting.