ZoneMinder/zoneminder #3806: v1.37.50: continuous event only lasts the minimum section length or nothing records at all

AndrStnz, 2024-01-23:
**Describe Your Environment**
- Version of ZoneMinder v1.37.50
- How you installed ZoneMinder PPA
- Full name and version of OS Ubuntu 20.04.6 LTS

**Describe the bug**
The continuous events only last the minimum section length all of the sudden. (10Secs)
If I set the minimum to 60 seconds nothing is recorded at all... mhm.

**Expected behavior**
continuous events last the maximum section length, like before :)

connortechnology, 2024-01-23:
What is your value of Options -> Config -> EVENT_CLOSE_MODE?

Is this just recording, no analysing?

Are you up to date?  Changes were made yesterday.  I am not able to reproduce this behaviour here.

eRaid6, 2024-01-23:
@connortechnology I am having the same issue here, except I have worked around it by setting `Minimum Section length` to the same value as `Section length` and I do get recordings that way.

My `EVENT_CLOSE_MODE` is set to `idle`, I have never changed that so I suspect it's the default.

AndrStnz, 2024-01-23:
EVENT_CLOSE_MODE is set to idle

I updated yesterday.

It is just recording, no analysis.

I am currently unable to get anything to record at all.
@eRaid6 good to hear you were able to work around this behaviour! :) 

I will try to rebuild my setup with a fresh lxc container tomorrow.

AndrStnz, 2024-01-23:
So... completely fresh install on ubuntu 22.04 (LXC) and same behaviour. 10 Second long recordings.
If I set the minimum to anything other than 10 seconds it completely breaks and no new events are created. 
Even when I set the minimum back to 10 secs it does not recover.

I tried with multiple cameras.

So what logs should I gather? :D

commit 403c63af (2024-01-23): Use Duration instead of timestamp % section_length.  Shorter code and actually works.  Fixes #3806
closed 2024-01-23 by connortechnology