ZoneMinder/zoneminder #3807: v1.37.50-latest Continuous Events still "broken"

AndrStnz, 2024-01-25:
ZoneMinder v1.37.50 (with the latest fix, thank you for that :) )
How you installed ZoneMinder PPA
Full name and version of OS Ubuntu 22.04.6 LTS

Describe the bug:
Monitors seem to randomly not create continuous events. Or leave events open.
I have some monitors that do not open continuous events at all.

This is the event list of a cam that works partially:
[image]

I cloned and recreated the monitors multiple times.
This is on my fresh test install.

Expected behavior
continuous events just record continuously one event after the other.

connortechnology, 2024-01-25:
I believe you have crashing. The events with no Duration are an indicator that the zmc process crashed before finalising the event. Please check yours logs.

AndrStnz, 2024-01-25:
Oh, yes you are absolutely right. OOM kills.
So, but this is also new with 1.37.50... 
`zoneminder zmdc[10878]: INF ['zmc -m 9' crashed, signal 8]`

Well, this explains a lot.
```
Jan 25 09:20:12 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 09:57:22 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 10:06:54 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 11:02:18 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 12:14:53 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 13:26:39 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 14:42:39 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 15:55:22 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 17:07:00 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
Jan 25 18:18:00 zoneminder systemd[1]: zoneminder.service: A process of this unit has been killed by the OOM killer.
```

AndrStnz, 2024-01-26:
I just downgraded to 1.37.49~20240116.22-jammy and as far as I can tell everything works flawlessly again, even the monitors that just would no create an event started working instantaneously.

AndrStnz, 2024-01-27:
1.37.49~20240116.22-jammy works perfectly.

Guess when I switched back to .49 ;) 
[image]

Can I help with some log files? 
I can create a second instance with 1.37.50 for troubleshooting purposes, if that helps.

nophead, 2024-01-27:
There is an environment variable to set the library paths. Not at a
computer but something like OPENSCADLIBS.

nophead, 2024-01-27:
Sorry replied to the wrong email

baudneo, 2024-01-29:
Just as a minor addition, I just upgraded to latest commit (0e88d764f0bbab3b27732eeb9369e72b3c4d1fb3) today. ZM now has a very slow mem leak that wasn't there in .49 (Don't recall which commit specifically). I am usually hovering around 4-7GB RAM for days on end, after a reboot and 5.5 hours of commit 0e88d764f0bbab3b27732eeb9369e72b3c4d1fb3, I am at 14.5GB and climbing slowly about 2-3 GB an hour.

AndrStnz, 2024-02-19:
Seems stable as of v1.37.51.
Thank you for your hard work @connortechnology !

closed 2024-02-19 by AndrStnz