ZoneMinder/zoneminder #3809: Memory leak in zmc from commit: 0e88d76 (1.37.50)

baudneo, 2024-01-29:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
https://github.com/ZoneMinder/zoneminder/commit/0e88d764f0bbab3b27732eeb9369e72b3c4d1fb3
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
source
- Full name and version of OS
Ubuntu Jammy
- Browser name and version (if this is an issue with the web interface)
N/A

**Describe the bug**
Log warning message may be a red herring?

Slow memory leak in zmc proccesses. Unsure of the root cause but the monitors slowly eat RAM until I restart that specific monitor. `max image buffer size` is not set to 0 (unlimited).

I do keep getting the packet queue is full and analysis may not be keeping up error, which doesnt make sense to me as analysis for motion detection is not running on these 3 specific monitors that are throwing that warning log, those monitors are also not decoding. ZM has access to 16 cores from 2x Xeon Silver 4110 processors and the CPU usage hovers around 5-8%, so I dont think it is underpowered?

The monitors are substreams (2x 640x360 , 1x Full D1 PAL 720x576 ), using source type FFMPEG, streaming @ 15 FPS, have a max image buffer of 1000, keyframe value is 29, Analysis = None, Decoding = None, Recording = Always with camera passthrough. 

2 of the monitors are the same model of IP PoE camera but the 3rd is a doorbell camera from a different manufacturer. So that should rule out a HW issue? 

I am now testing a 2000 and 3000 max image buffer and it is throwing the same error, I will keep uping the value to see if it goes away. 1000 frames at 15 FPS is roughly 66 seconds, likewise, 2000 is 132 seconds, etc.

**To Reproduce**
Steps to reproduce the behavior:
Start zm and watch the system RAM slowly increase over time.

**Expected behavior**
No mem leak.

**Debug Logs**
```

1/29/24, 10:26:45 AM MST	zmc_m1	545855	WAR	You have set the max video packets in the queue to 3000. The queue is full. Either Analysis is not keeping up or your camera's keyframe interval 29 is larger than this setting.	zm_packetqueue.cpp	155

```

baudneo, 2024-01-29:
It seems maybe the log isnt a red herring. The 3 monitors throwing the warning log are the ones eating all the RAM.

connortechnology, 2024-01-29:
Def not a red herring.  3000 is a lot for the packetqueue.  

I fixed one memleak which may affect the camera going offline case.  So everyone please update and retest.

connortechnology, 2024-01-29:
Please note that the analysis thread is what consumes the packets from the queue and writes out events. So even not doing analysis, the thread is doing something, and for whatever reason is not keeping up.  I

baudneo, 2024-01-30:
After testing for a while, the mem leak is still there. I'll do some debug logging on one of the affected monitors.

baudneo, 2024-01-30:
Im not too sure what I should be looking for. Only thing I can really decipher is it seems the video packet queue is almost at 5000 and I am assuming the memory use is from that?

Is there anything I should do to try and drill down into whats causing it to lag behind?

```log
  01/29/24 17:36:26.399702 zmc_m9[145996].DB4-zm_ffmpeg_camera.cpp/224 [Using video input because 337094250 >= 337080100]
  01/29/24 17:36:26.420455 zmc_m9[145996].DB2-zm_ffmpeg_camera.cpp/252 [ffmpeg_camera in: pts: 2697077 * 1/8000=337.134625, dts: 2697077,
   size: 320, stream_index: 1, audio flags: 0001, keyframe(1) pos: -1, duration: 320]
  01/29/24 17:36:26.420517 zmc_m9[145996].DB4-zm_monitor.cpp/2585 [Back from capture result=1 image count 4981]
  01/29/24 17:36:26.420527 zmc_m9[145996].DB2-zm_monitor.cpp/2613 [Have packet stream_index:1 ?= videostream_id: 0 q.vpktcount 4967 event
  ? 0 image_count 4981]
  01/29/24 17:36:26.420534 zmc_m9[145996].DB2-zm_monitor.cpp/2637 [Queueing audio packet]
  01/29/24 17:36:26.420543 zmc_m9[145996].DB2-zm_packetqueue.cpp/142 [packet counts for 1 is 8278]
  01/29/24 17:36:26.420550 zmc_m9[145996].DB4-zm_packetqueue.cpp/216 [packetqueue queuepacket, unlocked signalling]
  01/29/24 17:36:26.420559 zmc_m9[145996].DB4-zm_ffmpeg_camera.cpp/224 [Using video input because 337134625 >= 337080100]
  01/29/24 17:36:26.422168 zmc_m9[146524].DB4-zm_packetqueue.cpp/466 [Locking in get_packet using it 0x560ca84a4360 queue end? 0]
  01/29/24 17:36:26.422260 zmc_m9[146524].DB4-zm_packetqueue.cpp/472 [Have Lock in get_packet]
  01/29/24 17:36:26.422284 zmc_m9[146524].DB3-zm_packetqueue.cpp/485 [get_packet using it 0x560ca84a4360 locking index 14]
  01/29/24 17:36:26.422307 zmc_m9[146524].DB4-zm_packet.h/110 [TryLocking packet 14 0x560ca8401c80 locked: 0 owns: 0]
  01/29/24 17:36:26.422328 zmc_m9[146524].DB4-zm_packet.h/112 [TryLocking packet 14 0x560ca8401c80 1, owns: 1]
  01/29/24 17:36:26.422348 zmc_m9[146524].DB2-zm_packetqueue.cpp/490 [Locked packet 14, unlocking packetqueue mutex]
  01/29/24 17:36:26.422367 zmc_m9[146524].DB4-zm_monitor.h/717 [Not ready because image_count(4981) <= ready_count(75)]
  01/29/24 17:36:26.422386 zmc_m9[146524].DB3-zm_monitor.cpp/1946 [Not ready?]
  01/29/24 17:36:26.422405 zmc_m9[146524].DB4-zm_packet.h/117 [packet 14 unlocked, 0x560ca8401c80, locked 1, owns 1]
  01/29/24 17:36:26.422424 zmc_m9[146524].DB4-zm_packet.h/120 [packet 14 unlocked, 0x560ca8401c80, locked 0, owns 0]
  01/29/24 17:36:26.422445 zmc_m9[146524].DB5-zm_analysis_thread.cpp/35 [Sleeping for 33333us]
  01/29/24 17:36:26.440352 zmc_m9[145996].DB2-zm_ffmpeg_camera.cpp/252 [ffmpeg_camera in: pts: 2697395 * 1/8000=337.174375, dts: 2697395,
   size: 320, stream_index: 1, audio flags: 0001, keyframe(1) pos: -1, duration: 320]
  01/29/24 17:36:26.440383 zmc_m9[145996].DB4-zm_monitor.cpp/2585 [Back from capture result=1 image count 4981]
  01/29/24 17:36:26.440392 zmc_m9[145996].DB2-zm_monitor.cpp/2613 [Have packet stream_index:1 ?= videostream_id: 0 q.vpktcount 4967 event
  ? 0 image_count 4981]
  01/29/24 17:36:26.440399 zmc_m9[145996].DB2-zm_monitor.cpp/2637 [Queueing audio packet]
  01/29/24 17:36:26.440406 zmc_m9[145996].DB2-zm_packetqueue.cpp/142 [packet counts for 1 is 8279]
  01/29/24 17:36:26.440414 zmc_m9[145996].DB4-zm_packetqueue.cpp/216 [packetqueue queuepacket, unlocked signalling]
  01/29/24 17:36:26.440432 zmc_m9[145996].DB4-zm_ffmpeg_camera.cpp/224 [Using video input because 337174375 >= 337080100]
  01/29/24 17:36:26.455923 zmc_m9[146524].DB4-zm_packetqueue.cpp/466 [Locking in get_packet using it 0x560ca84a4360 queue end? 0]
  01/29/24 17:36:26.455948 zmc_m9[146524].DB4-zm_packetqueue.cpp/472 [Have Lock in get_packet]
  01/29/24 17:36:26.455956 zmc_m9[146524].DB3-zm_packetqueue.cpp/485 [get_packet using it 0x560ca84a4360 locking index 14]
  01/29/24 17:36:26.455963 zmc_m9[146524].DB4-zm_packet.h/110 [TryLocking packet 14 0x560ca8401c80 locked: 0 owns: 0]
  01/29/24 17:36:26.455969 zmc_m9[146524].DB4-zm_packet.h/112 [TryLocking packet 14 0x560ca8401c80 1, owns: 1]
  01/29/24 17:36:26.455976 zmc_m9[146524].DB2-zm_packetqueue.cpp/490 [Locked packet 14, unlocking packetqueue mutex]
  01/29/24 17:36:26.455981 zmc_m9[146524].DB4-zm_monitor.h/717 [Not ready because image_count(4981) <= ready_count(75)]
  01/29/24 17:36:26.455987 zmc_m9[146524].DB3-zm_monitor.cpp/1946 [Not ready?]
  01/29/24 17:36:26.455992 zmc_m9[146524].DB4-zm_packet.h/117 [packet 14 unlocked, 0x560ca8401c80, locked 1, owns 1]
  01/29/24 17:36:26.455998 zmc_m9[146524].DB4-zm_packet.h/120 [packet 14 unlocked, 0x560ca8401c80, locked 0, owns 0]
  01/29/24 17:36:26.456004 zmc_m9[146524].DB5-zm_analysis_thread.cpp/35 [Sleeping for 33333us]
  01/29/24 17:36:26.460843 zmc_m9[145996].DB2-zm_ffmpeg_camera.cpp/252 [ffmpeg_camera in: pts: 30343209 * 1/90000=337.146767, dts: 303432
  09, size: 9764, stream_index: 0, video flags: 0000, keyframe(0) pos: -1, duration: 6000]
  01/29/24 17:36:26.460877 zmc_m9[145996].DB4-zm_monitor.cpp/2585 [Back from capture result=1 image count 4981]
  01/29/24 17:36:26.460886 zmc_m9[145996].DB2-zm_monitor.cpp/2613 [Have packet stream_index:0 ?= videostream_id: 0 q.vpktcount 4967 event
  ? 0 image_count 4981]
  01/29/24 17:36:26.460895 zmc_m9[145996].DB2-zm_packetqueue.cpp/142 [packet counts for 0 is 4968]
  01/29/24 17:36:26.460924 zmc_m9[145996].DB4-zm_packetqueue.cpp/216 [packetqueue queuepacket, unlocked signalling]
[... cut ...]
  01/29/24 17:36:26.690794 zmc_m9[146524].DB4-zm_monitor.h/717 [Not ready because image_count(4985) <= ready_count(75)]
  01/29/24 17:36:26.690800 zmc_m9[146524].DB3-zm_monitor.cpp/1946 [Not ready?]
  01/29/24 17:36:26.690807 zmc_m9[146524].DB4-zm_packet.h/117 [packet 14 unlocked, 0x560ca8401c80, locked 1, owns 1]
  01/29/24 17:36:26.690814 zmc_m9[146524].DB4-zm_packet.h/120 [packet 14 unlocked, 0x560ca8401c80, locked 0, owns 0]
  01/29/24 17:36:26.690821 zmc_m9[146524].DB5-zm_analysis_thread.cpp/35 [Sleeping for 33333us]
```

connortechnology, 2024-01-30:
Are you doing decoding=ondemand?  I think I know what the issue is.  Your analysis thread isn't doing anything because it is waiting for a certain number of frames to be decoded.

I should have a fix shortly.

baudneo, 2024-01-30:
On those monitors Decoding=None. No live view, just capture and record using camera passthrough

connortechnology, 2024-01-30:
Yeah same problem should be fixed by latest build.

baudneo, 2024-01-31:
That fixed it, stable around 5GB for a few hours now.

closed 2024-01-31 by baudneo