ZoneMinder/zoneminder #3810: Color MJPEG cv2 stream being decoded in grayscale sometimes

maddios, 2024-02-02:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.33
- How you installed ZoneMinder: docker image: ghcr.io/zoneminder-containers/zoneminder-base:1.36.33@sha256:8ee8031bc387fa8e67d10ce0202424c1b8cfc0dd6bada70aa237d1698ce28a2e
- Full name and version of OS: not 100% sure which os this container runs

**If the issue concerns a camera**
- Python cv2 mjpeg stream (opencv version 4.9.0.80)
- 5 fps
- 1920x1080, but happens in all the streams (i have a total of 5)
- ZoneMinder Source Type: ffmpeg

**Describe the bug**
I'm running a yolov8 python app which processes my camera stream then outputs it via cv2 using this command:

`cv2.imencode(".jpg", frame, [cv2.IMWRITE_JPEG_QUALITY, 50])`

chrome opens the stream just fine and it's always color, however in zoneminder when I restart the stream it'll be grayscale, to fix I have to restart my zm, sometimes restarting the stream fixes it but mostly not.

I've tried streaming these streams to a mjpg file via ffmpeg and it grabs them just fine in color
I've never had this issue with any other streams to zm, so I'm really confused what could cause it to go grayscale

**Debug Logs**
```

I'm trying to figure out which logs would be relevant to this bug, I tried dumping some debug logs but there's nothing in there regarding ffmpeg decode problems

```

baudneo, 2024-03-06:
This is not a zm issue but, I can help. Can you post a snippet of the code you use to ingest the stream? How do you pull images?

I have a custom ml setup and have no issues with zm being greyscale.

SteveGilvarry, 2024-09-13:
Closing as stale

closed 2024-09-13 by SteveGilvarry