ZoneMinder/zoneminder #3813: Memory leak in zmc during live viewing

IgorA100, 2024-02-04:
**Describe Your Environment**
- Version of ZoneMinder - 1.37.50
- How you installed ZoneMinder - PPA
- Full name and version of OS - Ubuntu 23.10 mantic

**If the issue concerns a camera**
_FIRST, ID=13 & THIRD, ID=6_ 
- frame rate - 15
- resolution: 3840x2160 H.264 (base)
- ZoneMinder Source Type:
Capturing - Always
Source Path - rtsp://192.168.111.20:554/main
Method - TCP
Options - - 
Decoding - On Demand
SourceSecondPath - rtsp://192.168.111.20:554/third (Yes, that's right, there are three streams on the camera!)
Decoder - Auto
DecoderHWAccelName - - 
DecoderHWAccelDevice - - 
Target colorspace - 32 bit color
Capture Resolution (pixels) - 1920x1080 1080p
Preserve Aspect Ratio - NO
Orientation - Normal
Deinterlacing - Disabled
Maximum FPS - - 
Alarm Maximum FPS - - 
Motion Detection - Always
Analysis FPS - 10,00
Recording - Always
Video Writer - Camera Passthrough

_SECOND, ID=9_ 
- frame rate - 15
- resolution: 2560x1440 H.264 (base)
- ZoneMinder Source Type:
Capturing - Always
Source Path - rtsp://192.168.111.111:80/0
Method - TCP
Options - - 
Decoding - On Demand
SourceSecondPath - rtsp://192.168.111.111:80/1
Decoder - Auto
DecoderHWAccelName - - 
DecoderHWAccelDevice - - 
Target colorspace - 32 bit color
Capture Resolution (pixels) - 1280x720 720p
Preserve Aspect Ratio - NO
Orientation - Normal
Deinterlacing - Disabled
Maximum FPS - - 
Alarm Maximum FPS - - 
Motion Detection - None
Analysis FPS - -
Recording - Always
Video Writer - Camera Passthrough

**Describe the bug**
During live viewing, all memory is consumed, including swap.
The effect of memory consumption on an 8k stream is especially noticeable

**To Reproduce**
Steps to reproduce the behavior:

1. Reboot the server.
2. Enable live viewing of 1 camera and watch for 5 minutes
3. Switch to live view of the second camera and watch for 5 minutes
4. Switch to live view of the third camera and watch for 5 minutes

**Expected behavior**
After switching to viewing the second (subsequent) camera, the memory used by the first (previous) camera must be cleared.

**Debug Logs**
```
There is nothing interesting
```
1 Reboot
[image]

2 Start viewing 1 monitor (m 13)
[image]

3 Continued viewing of 1 monitor (m 13), memory is actively being consumed
[image]

4 Continued viewing of 1 monitor (m 13), memory is actively being consumed. Swap is used
[image]

5 The viewing is over, the memory has not been freed.
[image]

6 Start viewing 2 monitor (m 9)
[image]

7 The viewing is over, the memory has not been freed
[image]

8 Start viewing 3 monitor (m 6)
[image]

9 Continued viewing of 3 monitor (m 6), memory is actively being consumed
[image]

10 Viewing 1 monitor again (m 13)
[image]

IgorA100, 2024-02-04:
Probably. The problem is more global than freeing memory by the process.
I tried setting Capture Resolution (pixels) - 640x360 for a camera outputting a 3840x2160 stream and the memory consumption of the "zmc -m 6" stream stopped at 4.6GB !!!. It seems to me that this is a lot.
Is this problem solvable or does it all come down to "ffmpeg" and there is no solution?
I know about "Janus". I'm trying to use it, but so far there are problems with the stream breaking.

connortechnology, 2024-02-06:
Please see all the other issues and threads about this very same topic. 

You are using ondemand decoding which means that when no one is viewing, it is doing almost nothing and not consuming memory. When you start viewing, it starts decoding, meaning that it is allocating space for a raw rgba image for each frame.  If your system can't keep up it will consume all ram. Check your MaxImageBUffers.  

Also, the SourceSecondPath does not do what you think it does.  It is specifically for an audio stream that comes from a second source.  Not for another video stream.. We may someday use it for that, but not at this time.

IgorA100, 2024-02-07:
Thanks for the answer.
I read a lot here and on the forum, I studied the documentation (but it is very outdated...) before writing.
During this testing, I did not have a limit on the maximum buffer size, because... They wrote on the forum about problems when using buffer limits. I hoped that the system itself would work correctly. In principle, the system works correctly, with the exception of the moment described in more detail below:

Today I tested the system once again and it seems that I found the reason for the memory overflow in the following case:
Available memory - 20GB
1. "Decoding =On Demand"
2. Start live viewing of monitor 1 and watch for a couple of minutes, waiting until the process consumes more than 10-12GB of memory
3. Stop live viewing (logically, decoding should stop)
4. Start live viewing of monitor 2 and...
4.1. See that the "zmc -m 2" process consumes more than 10GB
4.2. See that the "zmc -m 1" process STILL consumes more than 10GB, although viewing has been completed for a long time and the memory is already going to swap, and the "zmc -m 1" process does not clear it!!!
Even if I set "Maximum Image Buffer Size (frames) = 70", the memory usage situation does not improve.

This situation with incomprehensible memory consumption and not clearing it occurs if a large value is specified in the setting:
"Recording"->"Optional Encoder Parameters: reorder_queue_size=1000"
I probably put one extra zero...
Very strange and unpredictable behavior of the "reorder_queue_size=1000" parameter during live viewing, because If you don’t view it online, then memory consumption with this parameter indicated is normal, but during live viewing, memory consumption goes through the roof and it is not cleared on TIME.

Regarding "SourceSecondPath".
1. In this case, the second thread does not affect the memory problem at all. I tried removing the link to it.
2. I read that it is used only for analyzing the audio track, but... in the options there is a choice for recording and for motion analysis, then I thought that the second stream was already fully functioning. Moreover, it still somehow seems to work :)
Well, let's wait, or if I have time to figure all this out, I'll try to study the source code and contribute.

IgorA100, 2024-02-09:
Possibly an indirect confirmation of not clearing memory:
In the logs:
"WAR You have set the max video packets in the queue to 121. The queue is full. Either Analysis is not keeping up or your camera's keyframe interval 51 is larger than this setting. zm_packetqueue.cpp 155"
In this case, the camera is set to keyframe = 50

moorsey, 2024-02-15:
I can confirm there is a new issue here with a memory leak, recent PPA update, similar to this.

Have three servers, 160ish GB RAM each.  I see this rapidly filling until a full system crash, only since recent PPA update, no new cameras or other changes

moorsey, 2024-02-15:
Tested changing all cameras to keyframes only for decoding, so remove any live view triggering processing etc, but same issues experienced

connortechnology, 2024-02-15:
There was a serious unexpected side effect of changes made last night.  New packages should be available that fix it.

IgorA100, 2024-02-18:
Yesterday I updated to 1.37.51 from PPA
But oh, oh, it didn’t get any better.
Resolution 3840*2160, 20 FPS there are warnings:

You have set the max video packets in the queue to 371. The queue is full. Either Analysis is not keeping up or your camera's keyframe interval 22 is larger than this setting. zm_packetqueue.cpp 155

I tried "Maximum Image Buffer Size" and 121 and 221 and even 371.
Apparently the packages are stuck somewhere.

IgorA100, 2024-02-22:
The latest changes seem to have had a significant impact on reducing memory leaks.
I no longer observe catastrophic excess memory consumption.
I see that the memory is being cleared.

closed 2024-02-22 by IgorA100