ZoneMinder/zoneminder #3814: After deleting Options > Storage > storage ID 1, the disk space is still in uses

olignyf, 2024-02-04:
**Describe Your Environment**
- Version of ZoneMinder: v1.36.33
- Debian 12

**Describe the bug**
- Events in/ var/cache/zoneminder/events takes about 800GB.
- No easy way to remove or reduce it.

**To Reproduce**
- Click Options > Storage
- Check the checkbox of the only storage item
- Click "Delete"
- No space is freed on the system

**Expected behavior**
Delete storage should delete the storage on the filesystem

ovargasp, 2024-02-16:
olignyf, that's an expected behavior, what the "options -> storage" does is present Zoneminder with the files storage location (configure it for ZM usage), so when you delete the storage what you're actually doing is telling Zoneminder to stop using it, not to empty it... if you want to release disk what you need to do is delete events stored there... there are several options: use a filter (that's what's normally used to maintain disk space); OS commands to delete files, the UI (slowest way but allows you to control what you want to keep)

olignyf, 2024-05-10:
Surely a better UX would be beneficial :
- Optional checkbox offered to delete the files as well for people like me who want to purge the disk as well.
- Better wording to indicate that it doesn't "Delete" but "Clear" storage configuration or something appropriate.