ZoneMinder/zoneminder #3815: Source Second Path (v1.37.50 ) Stream reading problems

IgorA100, 2024-02-05:
**Describe Your Environment**
- Version of ZoneMinder - 1.37.50
- How you installed ZoneMinder - PPA (Updated 03 feb 24)
- Full name and version of OS - Ubuntu 23.10 mantic

**If the issue concerns a camera**
- Make and Model - Beward SV5017RBZ
- Source Path - rtsp://192.168.111.20:554/main
- Method - TCP
- Decoding - On Demand
- SourceSecondPath - rtsp://admin:admin@192.168.111.20:554/third
- Motion Detection -None

**Describe the bug**
The second stream is not readable.
At the same time, if I change the first stream with the second in the monitor settings, then the second stream (which was previously the first and was read without problems) is not readable again.
If for the second thread I specify:
SourceSecondPath - rtsp://192.168.111.20:554/third
A 401 authorization error follows.
It's likely that different algorithms are used to start reading the first and second streams.

**Expected behavior**
1. The second stream must be read.
2. It is also not clear whether it is necessary to specify a name and password in the second stream.
On my other camera model the second thread is listed as:
SourceSecondPath - rtsp://192.168.111.25:80/1
and it works!

<details>

<summary>Debug Logs</summary>

### **Debug Logs for rtsp://admin:admin@192.168.111.20:554/third**
```

02/05/24 14:39:44.811734 zmc_m6[15941].DB1-zm_logger.cpp/225 [LogOpts: level=DB3 effective=DB3, screen=OFF, database=INF, logfile=DB3->/var/log/zm/_debug.15941, syslog=INF]
02/05/24 14:39:44.811925 zmc_m6[15941].DB1-zm_utils.cpp/231 [Detected a x86\x86-64 processor with AVX]
02/05/24 14:39:44.813329 zmc_m6[15941].DB2-zm_storage.cpp/61 [Loading Storage for 4 using SELECT `Id`, `Name`, `Path`, `Type`, `Scheme` FROM `Storage` WHERE `Id`=4]
02/05/24 14:39:44.813607 zmc_m6[15941].DB1-zm_storage.cpp/79 [Loaded Storage area 4 'NewStorageBIG']
02/05/24 14:39:44.813635 zmc_m6[15941].DB1-zm_monitor.cpp/397 [Have camera type Ffmpeg]
02/05/24 14:39:44.813667 zmc_m6[15941].DB3-zm_monitor.cpp/489 [Decoding: 2 savejpegs 0 videowriter 2]
02/05/24 14:39:44.813688 zmc_m6[15941].DB1-zm_packetqueue.cpp/739 [Setting pre_event_video_packet_count to 5]
02/05/24 14:39:44.813702 zmc_m6[15941].DB1-zm_packetqueue.cpp/733 [Setting max_video_packet_count to 121]
02/05/24 14:39:44.813739 zmc_m6[15941].DB2-zm_camera.cpp/69 [New camera id: 6 width: 640 line size: 2560 height: 360 colours: 4 subpixelorder: 8 capture: 1, size: 921600]
02/05/24 14:39:44.813783 zmc_m6[15941].INF-zm_ffmpeg.cpp/83 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
02/05/24 14:39:44.814039 zmc_m6[15941].DB2-zmc.cpp/218 [1 monitors loaded]
02/05/24 14:39:44.814063 zmc_m6[15941].INF-zmc.cpp/221 [Starting Capture version 1.37.50]
02/05/24 14:39:44.814128 zmc_m6[15941].DB3-zm_monitor.cpp/2452 [Reloading zones for monitor Camera_8Mp have 0]
02/05/24 14:39:44.816033 zmc_m6[15956].DB1-zm_db.cpp/205 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1707133184.813783, 'zmc_m6', 0, 15941, 0, 'INF', 'Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options', 'zm_ffmpeg.cpp', 83 ), thread_id: 2746]
02/05/24 14:39:44.817939 zmc_m6[15956].DB1-zm_db.cpp/205 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1707133184.814063, 'zmc_m6', 0, 15941, 0, 'INF', 'Starting Capture version 1.37.50', 'zmc.cpp', 221 ), thread_id: 2746]
02/05/24 14:39:44.818477 zmc_m6[15941].DB1-zm_zone.cpp/844 [Got 1 zones for monitor Camera_8Mp]
02/05/24 14:39:44.818505 zmc_m6[15941].DB3-zm_zone.cpp/770 [Got coordinate 0,0 from polygon string]
02/05/24 14:39:44.818520 zmc_m6[15941].DB3-zm_zone.cpp/770 [Got coordinate 639,0 from polygon string]
02/05/24 14:39:44.818532 zmc_m6[15941].DB3-zm_zone.cpp/770 [Got coordinate 639,359 from polygon string]
02/05/24 14:39:44.818545 zmc_m6[15941].DB3-zm_zone.cpp/770 [Got coordinate 0,359 from polygon string]
02/05/24 14:39:44.818558 zmc_m6[15941].DB3-zm_zone.cpp/782 [Successfully parsed polygon string 0,359]
02/05/24 14:39:44.819415 zmc_m6[15941].DB1-zm_monitor.cpp/2454 [Reloading zones for monitor Camera_8Mp have 1]
02/05/24 14:39:44.819443 zmc_m6[15941].DB1-zm_monitor.cpp/855 [SharedData=856 TriggerData=560 zone_count 1 * sizeof int 4 VideoStoreData=4128 timestamps=80 images=5x921600 = 4608000 analysis images=5x921600 = 4608000 image_format = 5x4 = 20 total=9221712]
02/05/24 14:39:44.819488 zmc_m6[15941].DB3-zm_monitor.cpp/888 [Success opening mmap file at (/dev/shm/zm.mmap.6)]
02/05/24 14:39:44.819517 zmc_m6[15941].DB3-zm_monitor.cpp/921 [MMap file size is 0]
02/05/24 14:39:44.834562 zmc_m6[15941].DB3-zm_monitor.cpp/964 [Aligning shared memory images to the next 64 byte boundary 0x7fea2752d5fc to 0x7fea2752d600 moved 4 bytes]
02/05/24 14:39:44.836285 zmc_m6[15941].DB1-zm_monitor.cpp/2460 [Reloading linked monitors for monitor Camera_8Mp, '']
02/05/24 14:39:44.836322 zmc_m6[15941].DB1-zm_monitor.cpp/1147 [Not Starting ONVIF]
02/05/24 14:39:44.836341 zmc_m6[15941].DB3-zm_monitor.cpp/1168 [Success connecting]
02/05/24 14:39:44.838714 zmc_m6[15941].DB1-zm_db.cpp/205 [Success running sql query INSERT INTO Monitor_Status (MonitorId,Status,CaptureFPS,AnalysisFPS,CaptureBandwidth) VALUES (6, 'Running',0,0,0) ON DUPLICATE KEY UPDATE Status='Running',CaptureFPS=0,AnalysisFPS=0,CaptureBandwidth=0, thread_id: 2746]
02/05/24 14:39:44.838764 zmc_m6[15941].DB1-zm_ffmpeg_camera.cpp/177 [Priming capture from rtsp://192.168.111.20:554/main]
02/05/24 14:39:44.838799 zmc_m6[15941].DB1-zm_ffmpeg_camera.cpp/344 [Calling avformat_open_input for rtsp://192.168.111.20:554/main]
02/05/24 14:39:44.838858 zmc_m6[15941].DB1-zm_ffmpeg_camera.cpp/354 [Rebuilt URI with encoded parameters: 'rtsp://admin:admin@192.168.111.20:554/main']
02/05/24 14:39:44.839061 zmc_m6[15941].DB3-zm_ffmpeg.cpp/68 [No default whitelist set]
02/05/24 14:39:44.839106 zmc_m6[15941].DB3-zm_ffmpeg.cpp/68 [Original list of addresses:]
[... 332 lines cut ...]
02/05/24 14:39:49.662108 zmc_m6[16020].DB2-zm_event.cpp/316 [Adding to event: pts: 237600, dts: 237600, size: 951097, stream_index: 0, flags: 0001, keyframe(1) pos: -1, duration: 3600]
02/05/24 14:39:49.662134 zmc_m6[16020].DB1-zm_videostore.cpp/1031 [Queue size for 0 is 0]
02/05/24 14:39:49.662150 zmc_m6[16020].DB1-zm_videostore.cpp/1060 [Pushing on queue 0, size is 1]
02/05/24 14:39:49.662166 zmc_m6[16020].DB1-zm_event.cpp/402 [Have frame type Normal from score(0) state 1 frames 1 bulk frame interval 100 and mod1]
02/05/24 14:39:49.662203 zmc_m6[16020].DB1-zm_event.cpp/456 [No image]
02/05/24 14:39:49.662218 zmc_m6[16020].DB1-zm_event.cpp/473 [Frame delta is 1707133189.65 s - 1707133189.65 s = 0.00 s, score 0 zone_stats.size 0]
02/05/24 14:39:49.662236 zmc_m6[16020].DB1-zm_event.cpp/506 [Not Adding 1 frames to DB because write_to_db:0 or frames > analysis fps 0.000000 or BULK]
02/05/24 14:39:49.662250 zmc_m6[16020].DB1-zm_event.cpp/694 [Adding packet 4]
02/05/24 14:39:49.662264 zmc_m6[16020].DB2-zm_event.cpp/314 [have_video_keyframe 1 codec_type 0 == video? 1 packet keyframe 0]
02/05/24 14:39:49.662279 zmc_m6[16020].DB2-zm_event.cpp/316 [Adding to event: pts: 248400, dts: 248400, size: 56952, stream_index: 0, flags: 0000, keyframe(0) pos: -1, duration: 3600]
02/05/24 14:39:49.662291 zmc_m6[16020].DB1-zm_videostore.cpp/1031 [Queue size for 0 is 1]
02/05/24 14:39:49.662306 zmc_m6[16020].DB1-zm_videostore.cpp/1041 [Found in order packet]
02/05/24 14:39:49.662320 zmc_m6[16020].DB1-zm_videostore.cpp/1060 [Pushing on queue 0, size is 2]
02/05/24 14:39:49.662339 zmc_m6[16020].DB1-zm_event.cpp/402 [Have frame type Normal from score(0) state 1 frames 2 bulk frame interval 100 and mod2]
02/05/24 14:39:49.662354 zmc_m6[16020].DB1-zm_event.cpp/456 [No image]
02/05/24 14:39:49.662375 zmc_m6[16020].DB1-zm_event.cpp/694 [Adding packet 5]
02/05/24 14:39:49.662392 zmc_m6[16020].DB2-zm_event.cpp/314 [have_video_keyframe 1 codec_type 0 == video? 1 packet keyframe 0]
02/05/24 14:39:49.662409 zmc_m6[16020].DB2-zm_event.cpp/316 [Adding to event: pts: 252000, dts: 252000, size: 43202, stream_index: 0, flags: 0000, keyframe(0) pos: -1, duration: 3600]
02/05/24 14:39:49.662430 zmc_m6[16020].DB1-zm_videostore.cpp/1031 [Queue size for 0 is 2]
02/05/24 14:39:49.662447 zmc_m6[16020].DB1-zm_videostore.cpp/1041 [Found in order packet]
02/05/24 14:39:49.662465 zmc_m6[16020].DB1-zm_videostore.cpp/1060 [Pushing on queue 0, size is 3]
02/05/24 14:39:49.662482 zmc_m6[16020].DB1-zm_event.cpp/402 [Have frame type Normal from score(0) state 1 frames 3 bulk frame interval 100 and mod3]

```
</details>

IgorA100, 2024-02-06:
I checked the operation of the second thread (used for motion detection), and what’s most surprising is that it seems to be working despite errors in the logs:
_"Failed to open secondary input"
"Could not find input codec"
"Unknown stream type"_
SourceSecondPath line: rtsp://admin:admin@192.168.111.20:554/third
Detection frames are saved. But I did not study the detection in detail. I think there are problems with the timeline there too.

IgorA100, 2024-04-12:
@connortechnology 
I can't understand whether the second thread is used when detecting motion or not?
It seems that analysis is taking place and events are being recorded.
But the processor load on the camera is the same when analyzing the first stream of 3840 x 2160 and when analyzing the second stream of 1280 x 720.

connortechnology, 2024-04-12:
The second path is for audio only.  It is an incomplete feature.

SteveGilvarry, 2024-09-13:
Closing as working as designed currently.

closed 2024-09-13 by SteveGilvarry