ZoneMinder/zoneminder #3816: Unable to connect to MySQL 8.0.35 server over SSL

jmpalacios, 2024-02-05:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.33.
- How you installed ZoneMinder: FreeBSD package.
- Full name and version of OS: FreeBSD 13.2 (jail).
- PHP version: 8.2.14

**Describe the bug**
Zoneminder's PHP code is unable to connect to a MySQL sever over SSL with a self-signed CA database server certificate, producing the error "SQLSTATE[HY000] [2002] Cannot connect to MySQL using SSL".

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Zoneminder's `ZM_DB_SSL_CA_CERT`, `ZM_DB_SSL_CLIENT_KEY`, & `ZM_DB_SSL_CLIENT_CERT` parameters, pointing them at the self-signed CA certificate, and client certificate and key files.
2. Start Zoneminder's PHP-FPM server.
3. Load Zoneminder in the browser.
4. Confirm the error stated above.

**Expected behavior**
Zoneminder should be able to connect to a database server with a self-signed CA certificate.

The problem is rather trivial to solve by editing the `includes/database.php` file and inserting the `PDO::MYSQL_ATTR_SSL_VERIFY_SERVER_CERT => false` PDO configuration option right below the `PDO::MYSQL_ATTR_SSL_CERT` option. Using that, Zoneminder can easily connect to the database server and the error goes away. So, perhaps the server CA certificate verification setting can be abstracted away behind a new `ZM_DB_SSL_VERIFY_SERVER_CERT` parameter, or something similar?

Please let me know if you'd like a merge request for something like this.

Than you!

connortechnology, 2024-02-05:
By all means, send a PR.  Yup, let's add another ZM config for that.

jmpalacios, 2024-02-05:
> By all means, send a PR. Yup, let's add another ZM config for that.

Sure thing, will do so ASAP!

jmpalacios, 2024-02-06:
I've submitted PR 3817 to take care of this issue. Please keep in mind that I'm far from familiar with the ZoneMinder source, so I took my best shot at covering what sifting through the code suggested was all the necessary bases.

Please let me know if I missed anything, thank you!

SteveGilvarry, 2024-09-13:
#3817 Merged, closing issue

connortechnology, 2024-09-13:
I had to revert #3817 so this should stay open.  Whatever the issue was is probably easy to solve I just never got back to it.