ZoneMinder/zoneminder #3836: v1.37.51 02.21.2024 The functions $thisServer->CpuUsagePercent() and $this Server->Cpu Load() are missing !!!

IgorA100, 2024-02-21:


connortechnology, 2024-02-21:
This is a useless issue report.  There is no reason for you to care about them.  Please explain how/why you care about these.

IgorA100, 2024-02-22:
Oh, Isaac, I apologize for not describing the situation in detail, because... I thought the problem was obvious.
After updating from PPA, PHP throws an error:
```
[Wed Feb 21 23:16:16.976261 2024] [php:error] [pid 406211] [client 192.168.111.77:64967] PHP Fatal error:  Uncaught Error: Call to a member function CpuLoad() on null in /usr/share/zoneminder/www/skins/classic/includes/functions.php:410\nStack trace:\n#0 /usr/share/zoneminder/www/ajax/status.php(21): getSysLoadHTML()\n#1 /usr/share/zoneminder/www/index.php(244): require_once('...')\n#2 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/includes/functions.php on line 410, referer: http://192.168.111.244/zm/index.php?view=watch&mid=6
[Wed Feb 21 23:16:16.989745 2024] [php:error] [pid 406246] [client 192.168.111.77:64961] PHP Fatal error:  Uncaught Error: Call to a member function CpuLoad() on null in /usr/share/zoneminder/www/skins/classic/includes/functions.php:410\nStack trace:\n#0 /usr/share/zoneminder/www/ajax/status.php(21): getSysLoadHTML()\n#1 /usr/share/zoneminder/www/index.php(244): require_once('...')\n#2 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/includes/functions.php on line 410, referer: http://192.168.111.244/zm/index.php?view=log
[Wed Feb 21 23:17:00.176529 2024] [php:error] [pid 406203] [client 192.168.111.252:56344] PHP Fatal error:  Uncaught Error: Call to a member function CpuUsagePercent() on null in /usr/share/zoneminder/www/skins/classic/includes/functions.php:396\nStack trace:\n#0 /usr/share/zoneminder/www/skins/classic/includes/functions.php(275): getCpuUsageHTML()\n#1 /usr/share/zoneminder/www/skins/classic/includes/functions.php(186): getNormalNavBarHTML()\n#2 /usr/share/zoneminder/www/skins/classic/views/console.php(99): getNavBarHTML()\n#3 /usr/share/zoneminder/www/index.php(260): require_once('...')\n#4 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/includes/functions.php on line 396
```
I quickly looked at the code and really did not find the CpuUsagePercent() and CpuLoad() functions being called, which should implement the new functionality for displaying CPU load.

commit 1742c03b (2024-02-22): Handle non-multi-server case. Fixes #3836
closed 2024-02-22 by connortechnology