ZoneMinder/zoneminder #3854: Events all report continuous even when it is a motion event.

baudneo, 2024-03-06:
My custom ml setup checks the cause of the event, ie. Motion|Continuous|Forced web|ONVIF and doesn't run if it isn't a motion or triggered event (skips continuous). I don't know exactly which commit it was, but I compiled on march 2nd around 10pm MDT.

Not a big issue, just thought I would mention it.

connortechnology, 2024-03-07:
I completely refactored the state machine logic...  and I've certainly seen similar reports, but thought it resolved in late Feb.  Need debug logs around the start of an event that should have been cause=Motion.  Also there are new EVENT_CLOSE_MODE settings that may be the issue.  It may be incorrectly falling bcak to IDLE instead of Alarm.

baudneo, 2024-03-07:
Kk, I'll get some debug logs this evening. Thanks!

Simpler1, 2024-03-08:
I'm still seeing this problem also.  About 80% of my events show a cause of Continuous even though I only ever record on motion.

baudneo, 2024-03-09:
Log file is huge lol, I think the event starts somewhere around 18:21:22 ish. This monitor was a view only monitor, no recording or modect. I turned modect on to create an event and it came out as continuous, so the logging should hopefully show what's up.

[zmc_m15.log](https://github.com/ZoneMinder/zoneminder/files/14544521/zmc_m15.log)

connortechnology, 2024-03-13:
Latest code may fix this. Please update and report.

ppnacho69, 2024-03-14:
I'm at V1.37.55 updated yesterday and I'm also full of Continuous "Cause" events. 
[image]

connortechnology, 2024-03-14:
I believe 503b1ae97 fixes this.

ppnacho69, 2024-03-14:
Runs nice now, only Motion events are recorded. But now when alarmed, the red border of the image doesn't appear, it keeps being white as on idle state. I'm running on V1.37.55) upgraded today.

ppnacho69, 2024-03-14:
Solved, only event triggers are recorded now on V1.37.55. Thank you

baudneo, 2024-03-15:
Fixed!

closed 2024-03-15 by baudneo