ZoneMinder/zoneminder #3858: Some 'Analysis' monitors not creating motion events, unknown why.

baudneo, 2024-03-09:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
https://github.com/ZoneMinder/zoneminder/commit/f0088e49d7a5d3464c8fa9e50e65369ad4645cbb
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
source
- Full name and version of OS
Ubuntu Jammy
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
Only some (2/5) Monitors set as: 
- Capture: always 
- Decoding: always
- Analysis: always
- Recording: on motion/trigger

are not triggering anymore and haven't since February 18th. Restarting the monitors or zm itself doesn't help. I did not change any monitor OR zone configs at all, and I have checked the DB to make sure the settings are indeed as the frontend says.

If I manually trigger an event, it works. The zone sensitivity is maxed to test. I remember having an issue like this before, the DB had different values than the frontend was reporting, but this isn't the case from what I can see this time. Kind of stumped as where to start. 

I may go through the front end and click save on zones and configs just to rewrite them to the DB and see if that helps, but as of now, I can't get 2 of my monitors to create motion events when they have been working flawlessly for quite a long time before Feb 18th.

I have compiled 2 times from source since Feb 18th and the issue persists through upgrades. Feb 18th commit also caused the other [issue](https://github.com/ZoneMinder/zoneminder/issues/3854) I have open with motion events being tagged as 'Continuous'. The commit for that is: https://github.com/ZoneMinder/zoneminder/commit/904f0627693538661c385933590af7315315dd81

**To Reproduce**
N/A

**Expected behaviour**
Events are created for monitors that are configured to.

**Debug Logs**
N/A

baudneo, 2024-03-11:
Did everything I could think of besides cloning the monitors and then deleting the old ones. I may just clone and delete but before I do, is there any debug logging I should be doing to see what the issue could be?

connortechnology, 2024-03-12:
As always the answers are in the logs.  I am surprised and worried that it isn't recording events.  If anything causes score to be non zero, we create an event.  Unless.. you have a strange or large value for alarm_frames meaning that you need a lot of alarmed frames before we create an event.

baudneo, 2024-03-12:
Just not sure where exactly I should be looking in debug logs. There's a large amount of output on zmc debug level 5. If you can give me a bit of direction to what lines or sections I should be homing into?

baudneo, 2024-03-12:
Kk thanks, I'll try drilling down tonight and report back.

rabsym, 2024-03-12:
I'm having the same issue as baudneo with latest master, going back solve issue.

connortechnology, 2024-03-13:
Latest code may fix this. Please update and report.

baudneo, 2024-03-13:
@rabsym what commit did u downgrade to? My issue started Feb 18th. 

Just updating, I haven't found logs of interest yet but, I did clone the monitors and the clones monitors are also not creating events. Just thought I should mention it.

I am @ work but will do an upgrade remotely over coffee break and report back.

rabsym, 2024-03-13:
@baudneo I was using zoneminder_1.37.50~20240126.68-bookworm_amd64.deb, i think commit is b5757a2d0553da58578a175c871279980bcf287c.

@connortechnology now opening events but still anything strange related to EVENT_CLOSE_MODE. What mode should i use to not doing continuous event but open when motion honoring "Alarm Frame Count" and closing when object_detected or going idle after alarm?

When first motion detected it start to do defined "section lenght" (5 minutes) events all time as seen in capture attached...

Thanks!

[image]

connortechnology, 2024-03-13:
I have a suspicion that somethen went wrong when I added a per-monitor event_close_mode setting on the Misc tab.  It should have defaulted to System, which means use the value from the Config table.  However, some of my monitors seem to have defaulted to "Time".  So double check that.  Your pic shows Time behaviour.

connortechnology, 2024-03-13:
Actually it shows the new Duration behaviour, which is really the old Time behaviour.

baudneo, 2024-03-13:
I am getting motion events on the cloned monitors now but they are labeled as continuous. Will test some more after work when I can go and wave my arms around in a zone.

rabsym, 2024-03-13:
I checked DB and it shows "idle" for this monitor because I changed it to try if that helped but 15 other monitors showing "system" and having the same behaviour.

baudneo, 2024-03-13:
Ok, so, monitors set to only record on motion are now recording continuous events. Motion detection appears to be working as continuous events also have motion notes.

rabsym, 2024-03-14:
Not sure because of still understanding states change but maybe related to this commit? https://github.com/ZoneMinder/zoneminder/commit/e78c8739448c6d2b13f08c6666d4f88f0d87b862

[image]

closed 2024-03-14 by connortechnology

baudneo, 2024-03-15:
Fixed.

commit 56fcd847 (2024-03-14): Fix motion events having cause continuous. Fixes #3858