ZoneMinder/zoneminder #3859: When deleting zones, server eventually returns 500 (possible mangled URL?)

baudneo, 2024-03-10:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
https://github.com/ZoneMinder/zoneminder/commit/f0088e49d7a5d3464c8fa9e50e65369ad4645cbb
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
source
- Full name and version of OS
Ubuntu Jammy
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
When I was editing zones, the server returned a 500 in the middle of the session. I was able to edit 1/3 zones then delete a zone. When I edited the last remaining zone (2/2 after delete) and went to save it, the server returned code 500 with no meaningful output in dev tools. The delete and save commands appear to have worked though.

When I tried to recreate the error, I could only do it when deleting a zone.

I did notice that the URL looks mangled: `http://10.0.1.7/zm/index.php?view=zones&mids[]=5`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new test zone in a monitor
2. mark and delete the new test zone
3. the 500 may return immediately or it may be after your next few requests.
4. See error

**Expected behavior**
No 500
**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2024-03-10:
There may be useful errors in apache2 or php_fpm logs depending on which web server you are running.Typically a 500 logs a syntax error something with some useful details.

connortechnology, 2024-03-10:
I think it may have to do with us assuming REQUEST['mids'] is set... quick fix incoming.

baudneo, 2024-03-10:
```
cat /var/log/apache2/error.log
[Sun Mar 10 00:00:04.115252 2024] [mpm_prefork:notice] [pid 142] AH00163: Apache/2.4.52 (Ubuntu) configured -- resuming normal operations
[Sun Mar 10 00:00:04.115301 2024] [core:notice] [pid 142] AH00094: Command line: '/usr/sbin/apache2'
[Sun Mar 10 08:21:40.303287 2024] [php:error] [pid 3313231] [client 10.0.1.10:40914] PHP Fatal error:  Uncaught Error: Call to undefined function arrap_map() in /usr/share/zoneminder/www/skins/classic/views/zones.php:26\nStack trace:\n#0 /usr/share/zoneminder/www/index.php(260): require_once()\n#1 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/views/zones.php on line 26, referer: http://10.0.1.7/zm/index.php?view=zones&mid=5
[Sun Mar 10 08:21:57.590226 2024] [php:error] [pid 3155750] [client 10.0.1.10:60966] PHP Fatal error:  Uncaught Error: Call to undefined function arrap_map() in /usr/share/zoneminder/www/skins/classic/views/zones.php:26\nStack trace:\n#0 /usr/share/zoneminder/www/index.php(260): require_once()\n#1 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/views/zones.php on line 26, referer: http://10.0.1.7/zm/index.php?view=zones&mid=5
[Sun Mar 10 08:22:09.373479 2024] [php:error] [pid 3155897] [client 10.0.1.10:35886] PHP Fatal error:  Uncaught Error: Call to undefined function arrap_map() in /usr/share/zoneminder/www/skins/classic/views/zones.php:26\nStack trace:\n#0 /usr/share/zoneminder/www/index.php(260): require_once()\n#1 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/views/zones.php on line 26
[Sun Mar 10 08:27:25.925641 2024] [php:error] [pid 3370322] [client 10.0.1.10:39712] PHP Fatal error:  Uncaught Error: Call to undefined function arrap_map() in /usr/share/zoneminder/www/skins/classic/views/zones.php:26\nStack trace:\n#0 /usr/share/zoneminder/www/index.php(260): require_once()\n#1 {main}\n  thrown in /usr/share/zoneminder/www/skins/classic/views/zones.php on line 26, referer: http://10.0.1.7/zm/?view=zones&mid=5
```

commit 3f9bda4f (2024-03-10): FIx arrap_map => array_map. Fixes #3859
commit b6472807 (2024-03-10): FIx arrap_map to array_map. Don't assume _REQUEST[mids] exists.  Fixes #3859
closed 2024-03-10 by connortechnology