ZoneMinder/zoneminder #3877: PTZ Controls layout could use improvement

connortechnology, 2024-03-19:
See here for a pic example:
https://forums.zoneminder.com/viewtopic.php?t=33148

I've never gotten around to improving them

connortechnology, 2024-03-19:
@IgorA100 maybe something you could tackle
@Simpler1 maybe something you might have an opinion on

IgorA100, 2024-03-19:
I was unable to reproduce this problem because... I don’t have this version of the browser at hand :(
But, as far as I understand, there are problems with specifying the size of the button display block.
As an option, I indicated on the forum what you can try to test my idea.
In 4-5 hours I will try to install Firefox 123.0.1 and check in more detail.
In general, the entire block of control buttons should probably be rewritten.

PR #3878 (2024-03-20): Fix: Location of PTZ Controls buttons when scaling browser less than 100%
Possible fix #3877