ZoneMinder/zoneminder #3887: Add support for DBD-MariaDB after DBD-mysql dropped support of the MariaDB

VVD, 2024-03-23:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.33
- How you installed ZoneMinder: build from ports
- Full name and version of OS: FreeBSD 13.3 amd64
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
A clear and concise description of what the bug is.
DBD-mysql dropped support of the MySQL 5.7 and MariaDB: https://github.com/perl5-dbi/DBD-mysql/issues/405#issuecomment-1818501464
If use MariaDB then can't use Zoneminder with recent DBD-mysql.
If install https://github.com/perl5-dbi/DBD-MariaDB instead then Zoneminder can't run.

**To Reproduce**
Steps to reproduce the behavior:
1. Replace DBD-mysql with DBD-MariaDB.
2. Start Zoneminder.
3. Zoneminder can't run.

**Expected behavior**
Run and work without errors with DBD-MariaDB.

**Debug Logs**
```
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
03/23/24 03:47:34.518328 zmpkg[25250].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 10) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
03/23/24 03:47:34.518977 zmpkg[25250].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 9) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
03/23/24 03:47:34.519803 zmpkg[25250].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 11) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
Unable to connect to DB. ZM Cannot continue.
BEGIN failed--compilation aborted at /usr/local/lib/perl5/site_perl/ZoneMinder/Config.pm line 150.
Compilation failed in require at /usr/local/lib/perl5/site_perl/ZoneMinder.pm line 33.
BEGIN failed--compilation aborted at /usr/local/lib/perl5/site_perl/ZoneMinder.pm line 33.
Compilation failed in require at /usr/local/bin/zmpkg.pl line 34.
BEGIN failed--compilation aborted at /usr/local/bin/zmpkg.pl line 34.
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
03/23/24 03:47:34.644731 zmpkg[25255].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 10) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
03/23/24 03:47:34.645274 zmpkg[25255].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 9) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
Use of uninitialized value $DBI::errstr in concatenation (.) or string at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 114.
03/23/24 03:47:34.645874 zmpkg[25255].ERR [ZoneMinder::Database:114] [Error reconnecting to db: errstr: error val:install_driver(mysql) failed: Can't locate DBD/mysql.pm in @INC (you may need to install the DBD::mysql module) (@INC contains: /usr/local/lib/perl5/site_perl/mach/5.36 /usr/local/lib/perl5/site_perl /usr/local/lib/perl5/5.36/mach /usr/local/lib/perl5/5.36) at (eval 11) line 3.
Perhaps the DBD::mysql perl module hasn't been fully installed,
or perhaps the capitalisation of 'mysql' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, MariaDB, Mem, Proxy, Sponge.
 at /usr/local/lib/perl5/site_perl/ZoneMinder/Database.pm line 110.]
Unable to connect to DB. ZM Cannot continue.
BEGIN failed--compilation aborted at /usr/local/lib/perl5/site_perl/ZoneMinder/Config.pm line 150.
Compilation failed in require at /usr/local/lib/perl5/site_perl/ZoneMinder.pm line 33.
BEGIN failed--compilation aborted at /usr/local/lib/perl5/site_perl/ZoneMinder.pm line 33.
Compilation failed in require at /usr/local/bin/zmpkg.pl line 34.
BEGIN failed--compilation aborted at /usr/local/bin/zmpkg.pl line 34.
```

connortechnology, 2024-03-28:
@abishai Thoughts?

abishai, 2024-03-28:
I'd say it needs testing. :p While the PR solves the problem with DBD-mysql, the more interesting question is it possible to build  ZM with MariaDB client libraries. 

But probably, this is not an issue as many Linux distros switched to MariaDB long time ago? I have MariaDB server, but clients are linked against mysql client.

connortechnology, 2024-03-28:
ubuntu 2310 here does not package dbd-mariadb. Debian bookworm does.  So it's not universal yet.  This change is needed for postgresql support as well.

VVD, 2024-03-28:
Zoneminder build fine with mariadb-client. But it can't run with p5-DBD-MariaDB.

SteveGilvarry, 2024-09-13:
For testing of 1.38

VVD, 2024-09-13:
> For testing of 1.38 @VVD can you build from master that has PR 3914 and confirm it works now.

I have only one host with zoneminder and it's production. So I can't test work, only build.

closed 2024-09-13 by connortechnology