ZoneMinder/zoneminder #3890: No response when pressing the "SET" button in PTZ control

IgorA100, 2024-03-23:
- 1.37.56~20240322163032

Нажатие кнопки 
`<button type="button" class="ptzTextBtn" value="Set" data-toggle="modal" data-target="#ctrlPresetModal"><?php echo translate('Set') ?></button>` 
skins\classic\includes\control_functions.php не обрабатывается в \skins\classic\views\js\watch.js

https://forums.zoneminder.com/viewtopic.php?p=133906#p133906

IgorA100, 2024-03-23:
data.html is empty, strange
```
function getCtrlPresetModal() {
  $j.getJSON(monitorUrl + '?request=modal&modal=controlpreset&mid=' + monitorId)
      .done(function(data) {
        insertModalHtml('ctrlPresetModal', data.html);
```

IgorA100, 2024-03-23:
The problem is here \ajax\modals\controlpreset.php:
if (!canEdit('Monitors')) return ' ';

connortechnology, 2024-03-23:
In a multi-server setup, we must pass authentication info along which we weren't doing.  Fixed by dd0b2bcf91a9be6537f311aff4a61b2b2556c7cf

closed 2024-03-23 by IgorA100