ZoneMinder/zoneminder #3912: Web interface stop working, no errors displayed, just a white page

ppnacho69, 2024-03-28:
**Environment**
- Ubuntu 20.04.6 LTS, 5.15.0-101-generic #111~20.04.1-Ubuntu 
- Browser version: Mozilla Firefox 124.0.1 (64-bit)
-  http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu

**Describe the bug**
After yesterday afternoon update, web screen is white, no errors or messages

**To Reproduce**
- Just try ton connect to http://127.0.0.1/zm or http://127.0.0.1/zm/index.php?view=console and nothing happens.

**Expected behavior**
Normal login or console page.

**Debug Logs Apache2 errors**
```
[Thu Mar 28 09:41:55.133653 2024] [php7:error] [pid 1014] [client 192.168.167.161:46212] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608, referer: http://192.168.167.206/zm/index.php?view=watch&mid=1
[Thu Mar 28 09:41:56.498368 2024] [php7:error] [pid 1015] [client 192.168.167.161:46226] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608, referer: http://192.168.167.206/zm/index.php?view=watch&mid=1
[Thu Mar 28 09:43:03.666315 2024] [php7:error] [pid 4973] [client 192.168.167.161:37314] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608
[Thu Mar 28 09:49:51.707314 2024] [php7:error] [pid 1011] [client 127.0.0.1:47754] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608
[Thu Mar 28 09:49:54.347756 2024] [php7:error] [pid 1013] [client 127.0.0.1:48532] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608
[Thu Mar 28 09:50:13.708483 2024] [php7:error] [pid 1015] [client 127.0.0.1:48366] PHP Parse error:  syntax error, unexpected '$this' (T_VARIABLE), expecting function (T_FUNCTION) or const (T_CONST) in /usr/share/zoneminder/www/includes/Monitor.php on line 608

```

connortechnology, 2024-03-28:
Was fixed last night. 97af155102cdb8d433eb2c049013ba67b1f55164

closed 2024-03-28 by connortechnology

ppnacho69, 2024-03-28:
Thank you very much